freifunk-gluon/gluon #733: Multithreadded compiling fails at high probability

MPW1412, 2016-04-23:
Hello,

I know, that many failures of compiling gluon are due to failed downloads. But I think that there are race conditions, too. As of my observations, the compiling process fails much more often, the more cores are used.

So I started collecting intel on this issue and I'm uploading two log files here now. I used my build script and enabled debugging, which is basically the "set -x" command in bash: https://github.com/FreiFunkMuenster/tools/blob/b66d89013feb6aff49b6c1ad51d440b2c1eadf3b/build_all.sh (Linked to fixed commit as work is in progress.)

Both tries failed because of uncompleted downloads.

[buildlogs.zip](https://github.com/freifunk-gluon/gluon/files/233182/buildlogs.zip)

Regards,
Matthias

christf, 2016-04-23:
we did discuss on irc to:
- [ ] cache downloads
- [ ] use more quotes
- [ ] get rid of some pipes
- [ ] order of builds (build all site.confs for one target first, then change target)

likely using a downlaod-cache will resolve your issues.

NeoRaider, 2016-04-24:
I think this is actually a race condition, caused by an incomplete backport. I'll fix this.

closed 2016-04-24 by NeoRaider

MPW1412, 2016-04-24:
Thanks for the quick fix. I'll do a dirclean and try to build master later.

MPW1412, 2016-04-30:
Works like a charm, just compiled in a brand new build environment with eight cores and it just worked.

commit 7f1c11cf (2016-04-24): Backport parallel download locking support

Fixes #733
commit 6e832e2b (2016-04-25): Backport parallel download locking support

Fixes #733