freifunk-gluon/gluon #744: Add Support for D-Link DIR-860L

mmalte, 2016-04-29:
This is a very interesting Device MediaTek MT7621AT with 880 MHz (Dual-Core), 16MB Flash, 128MB RAM, Dual Radio. ~75€

openWRT Support works very well:
https://wiki.openwrt.org/toh/d-link/dir-860l

We've reached around 480MBit/s via wifi without further tuning, running openwrt.

Logs:
[logread-dir-860l.txt](https://github.com/freifunk-gluon/gluon/files/243146/logread-dir-860l.txt)
[dmesg-dir-860l.txt](https://github.com/freifunk-gluon/gluon/files/243147/dmesg-dir-860l.txt)

Jotz is willing to provide devices:
https://forum.freifunk.net/t/suche-nach-zukuenftigen-gluon-dualband-geraet/11521/25

NeoRaider, 2016-07-03:
I've added support, but it doesn't work well yet. 11s has high packet loss, I haven't yet made further tests with the AP interface.

mmalte, 2016-07-04:
I've build a Firmware based on your patch. Hopefully @Ohnezahn sends us some Feedback.

http://ffac-build.skydisc.net/experimental/factory/gluon-ffac-2016.1.5-4~exp20160704-d-link-dir-860l-b1.bin
http://ffac-build.skydisc.net/experimental/sysupgrade/gluon-ffac-2016.1.5-4~exp20160704-d-link-dir-860l-b1-sysupgrade.bin

rotanid, 2017-08-07:
has someone tested this with LEDE-based gluon?
apparently there have been fixes to the driveres, maybe that solved the issues...
@mweinelt @mbaumga

mweinelt, 2017-08-07:
https://meshviewer.darmstadt.freifunk.net/#/de/map/908d7823af1c
Last online 4 days ago, so it's probably still loosing the 2.4 GHz Mesh connectivity, see https://github.com/freifunk-gluon/gluon/issues/1076.

rotanid, 2018-01-19:
closing this, as support is there. there's issue #1076 for the wifi problems.

closed 2018-01-19 by rotanid