freifunk-gluon/gluon #750: ath9k: "ath: phy0: Unable to reset channel, reset status -5"

mistersixt, 2016-05-03:
Moin moin,

ich habe eine Lokation beim FF-Wiesbaden, in der regelmässig (alle 1-2 Tage) ein TP-1043ND-v2 aussteigt mit:

ath: phy0: Unable to reset channel, reset status -5

Das war schon mit v2015.1.2 der Fall, und jetzt weiterhin auch mit der Version v2016.1.4. Ein Hardware-Austausch hat nichts gebracht. "wifi" bringt auch keinen Erfolg, nur ein Reboot hilft.

ath9k.ko Version ist 3.18.29.

ath9k/reset:

`    Baseband Hang:  7 `
 `Baseband Watchdog:  1 `
 `   Fatal HW Error:  0 `
 `      TX HW error:  0 `
 ` Transmit timeout:  0 `
   `   TX Path Hang:  3 `
  `     PLL RX Hang:  0 `
  `        MAC Hang:  0 `
  `    Stuck Beacon:  3 `
  `       MCI Reset:  0 `
 `Calibration error:  0 `
 `Tx DMA stop error:  1 `
 `Rx DMA stop error:  0 `

ath9k/queues:

 `(VO):  qnum: 3 qdepth:  0 ampdu-depth:  0 pending:   0 stopped: 0 `
 `(VI):  qnum: 2 qdepth:  0 ampdu-depth:  0 pending:   0 stopped: 0 `
 `(BE):  qnum: 1 qdepth:  3 ampdu-depth:  0 pending:   3 stopped: 0 `
 `(BK):  qnum: 0 qdepth:  0 ampdu-depth:  0 pending:   0 stopped: 0 `
 `(CAB): qnum: 8 qdepth:  0 ampdu-depth:  0 pending:   0 stopped: 0 `

Bin um jede Hilfe dankbar,

Gruss, mistersixt

NeoRaider, 2016-05-03:
Please try waiting 10~20s between `wifi down` and `wifi` to give the WLAN adapter some time to settle down.

Also, please make all bug reports in English, so we can forward them to the OpenWrt maintainers.

NeoRaider, 2016-05-03:
Potentially related upstream ticket: https://dev.openwrt.org/ticket/9654

NeoRaider, 2016-05-16:
We'll track all ath9k issues in #605. Please also try the updated branch *v2016.1.x-mac80211-test*.

closed 2016-05-16 by NeoRaider

shizhai, 2016-12-15:
Hi, I have got the same problem when the system boot:
`[   19.709363] ath: phy0: Unable to reset hardware; reset status -5 (freq 2412 MHz)
[   19.935336] ath: phy0: Unable to reset channel, reset status -5
[   19.941716] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   20.161466] ath: phy0: Unable to reset channel, reset status -5`

Do let me know if you need more information.