freifunk-gluon/gluon #751: l3: remove br-client, run radvd on every client interface

tcatm, 2016-05-03:
We'd like to remove br-client and also change the local-node feature a little:

- Client interfaces shouldn't be bridged anymore.
- Every client interface should have it's own local-node device (same IP and MAC).
- uradvd must run on all these local-node devices.
- l3roamd must be notified about new client interfaces.

christf, 2016-05-03:
translation from irc-discussion with tcatm: "it probably is best to duplicate the macvlan for br-client on all client interfaces. lateron uradvd may be startet on these"

NeoRaider, 2016-05-03:
Do we really need the macvlan? I don't see any reason to have interface-specific MAC addresses on client interfaces, so we could just set the MAC address of all client interfaces to the local-node address.

tcatm, 2016-05-03:
macvlan is not strictly required. If we have other means to ensure the anycast MAC works on these interfaces that's fine. I don't know if this works with wifi ap interfaces on Ralink.

belzebub40k, 2016-05-03:
Maybe its a good idea to reduce the use of virtual interfaces like macvlan if possible as they can cause errors as described in #680 . The crash is causes by platform independent code and can potentially happen on any device.

NeoRaider, 2016-05-03:
AP mode wifi is no problem, as such a MAC could as well lie behind a bridge, so the WLAN adapter can never filter any packets towards the master.

christf, 2016-05-04:
after some more discussion on irc with tcat we seem to have reached the following concept:
* macvlan is not used
* at first client-interfaces are defined in site.conf
* the client interfaces are put in a similar structure like gluon.mesh (gluon.client?) in pr747
* client interfaces are assigned the same mac and the same ipv6 address
* uradvd will be executed on all of them (the init script will handle starting uradvod on all client interfaces)
* l3roamd's init script will work out the client interfaces and hand them over to l3roamd

christf, 2017-02-15:
@NeoRaider do we still need this now that your macvlan/veth-work is merged?

NeoRaider, 2017-02-16:
We don't need it, but it is still something we might want to consider in the future.

mweinelt, 2020-03-03:
@NeoRaider Are we still considering this in any way?

NeoRaider, 2020-05-28:
@mweinelt I really don't know. It might be something to experiment on in the future, but I don't have any concrete plans regarding this matter.