freifunk-gluon/gluon #754: Wifi interfaces with vlans on 2016.1.4 are not added to bat0.

lcb01a, 2016-05-05:
Hi everyone,

While testing 2016.1.4 i've noticed that vlan interfaces ontop of the wifi interfaces are not added to bat0 anymore. This lead to a few test nodes which gone missing since the interface is not added.
Adding the interface manually works.

Here is a copy of the network config:

```
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd32:eb37:8029::/48'

config interface 'wan6'
        option proto 'dhcpv6'
        option ifname 'br-wan'
        option ip6table '1'
        option peerdns '0'
        option sourcefilter '0'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0 1 2 3 4'

config interface 'wan'
        option igmp_snooping '0'
        option ifname 'eth1'
        option auto '1'
        option peerdns '0'
        option type 'bridge'
        option proto 'dhcp'
        option macaddr 'a2:f4:c1:82:4d:0a'

config rule6 'wan6_lookup'
        option mark '0x01/0x01'
        option lookup '1'

config route6 'wan6_unreachable'
        option type 'unreachable'
        option table '1'
        option target '::/0'
        option metric '65535'
        option gateway '::'
        option interface 'loopback'

config interface 'client'
        option reqprefix 'no'
        option proto 'dhcpv6'
        option type 'bridge'
        option igmp_snooping '0'
        option macaddr 'a0:f3:c1:82:4d:0a'
        option peerdns '1'
        list ifname 'eth0'
        list ifname 'bat0'
        option sourcefilter '0'

config interface 'mesh_wan'
        option auto '0'
        option ifname 'br-wan'
        option mesh 'bat0'
        option proto 'batadv'

config interface 'mesh_lan'
        option ifname 'eth0'
        option auto '0'
        option macaddr 'a2:f4:c2:82:4d:0a'
        option mesh 'bat0'
        option proto 'batadv'

config interface 'mesh_vpn'
        option ifname 'mesh-vpn'
        option mesh_no_rebroadcast '1'
        option mesh 'bat0'
        option proto 'batadv'
        option mtu '1364'
        option macaddr 'a2:f7:c1:82:4d:0a'

config interface 'mesh_radio0_vlan'
        option ifname '@mesh_radio0.2'
        option mesh 'bat0'
        option proto 'batadv'
        option mtu '1532'

config interface 'bat0'
        option ifname 'bat0'
        option macaddr 'a0:f3:c1:82:4d:0a'
        option proto 'none'

config interface 'ibss_radio0'
        option proto 'none'
        option mtu '1536'

config interface 'ibss_radio0_vlan'
        option ifname '@ibss_radio0.2'
        option mesh 'bat0'
        option proto 'batadv'
        option mtu '1532'

config device 'local_node_dev'
        option macaddr '16:41:95:40:f7:dc'
        option ifname 'br-client'
        option name 'local-node'
        option type 'macvlan'

config interface 'local_node'
        option ifname 'local-node'
        option ipaddr '10.25.31.254'
        option ip6addr '2a03:2260:3004:25::ffff/128'
        option netmask '255.255.224.0'
        option proto 'static'

config route6 'local_node_route6'
        option target '2a03:2260:3004:25::/64'
        option gateway '::'
        option interface 'client'
```

NeoRaider, 2016-05-05:
Are the VLAN interfaces created automatically, and just not added to bat0 correctly? Can you test if this issue already occurred on Gluon 2016.1.3?

lcb01a, 2016-05-05:
This issue did not occur with 2016.1.3 as this is our current stable release of 2016.1.3.
The vlan interface itself is existing, just not present in bat0

NeoRaider, 2016-05-05:
Interesting, I'm not aware of a change between 2016.1.3 and 2016.1.4 that might have caused this... I'll look into this.

closed 2016-05-05 by NeoRaider

FreifunkUFO, 2016-05-07:
btw, with openwrt CC + DD trunk (via meshkit) we use a new config style for vlan-on-wifi:
[image]

NeoRaider, 2016-07-26:
I've pushed a new netifd update in aae05fe3683a640b022f775f1100060fc51cb8bd which should fix this issue, so I have removed the revert patch I had added before. It would be great if you could check if this is actually still fixed in the Gluon master.

commit 80d5f5f9 (2016-05-05): netifd: update and revert problematic patch

Fixes VLANs with shell protos on top of WLAN devices.

Fixes #754
commit 495a3ab9 (2016-05-05): netifd: update and revert problematic patch

Fixes VLANs with shell protos on top of WLAN devices.

Fixes #754