freifunk-gluon/gluon #757: Autoupdater kann ohne Mesh-DNS keine Hostnamen auflösen

RubenKelevra, 2016-05-09:
Soweit ich das sehe haben wir derzeit mit dem Autoupdater das Problem, dass er die falsche dnsmasq-Instanz benutzt um den DNS-Namen vom Update-Server aufzurufen.

Wir haben für "übers Internet" einen FQDN und zusätzlich eine IPv6 im Mesh als Update-Server definiert. Wenn aber der Knoten nicht mehr ins Mesh kommt funktioniert wegen fehlendem DNS-Server kein Autoupdate.

Meines Erachtens sollte der Autoupdater die WAN-dnsmasq-Instanz nach DNS-Lookup fragen. Mesh-Intern kann man ja mit Anycast-IPs arbeiten, oder interne IPs dafür eintragen.

NeoRaider, 2016-05-09:
Das ist "by design": die Regel, dass der WAN-Port für *nichts* benutzt wird außer dem VPN ist so ziemich die absoluteste Regel von Gluons Netzwerk-Config.

Neben dem Sicherheitsaspekt (wir können Menschen, die einen Knoten in ihr Netz stellen, genau sagen, was der Knoten im Netz tut) stellt das auch sicher, dass sich Knoten mit und ohne VPN nicht unterschiedlich verhalten.

closed 2016-05-09 by NeoRaider

RubenKelevra, 2016-05-09:
Ich weiß, dass das so gedacht ist, allerdings ist es unsinnig.

Es ist normal das man auf einem Webserver mehrere Domains laufen hat, auf einer IPv4 und entsprechend muss man den Updateserver mittels Domainnamen aufrufen. Ist der Knoten aber nicht mehr im Mesh kann er den Domainnamen nicht mehr aufrufen.

Damit ist die Chance bei Problemen oder inkompatiblen Updates den Updateserver noch via Client-WLAN-Connect zu erreichen gleich 0 außer man setzt einen Updateserver mit eigener IPv4 auf.

Im Mesh dagegen kann man beliebig viele IPs vorhalten, somit ist hier ein Domainname relativ gesehen unnötig im Gegensatz zum "im Internet stehenden" Server.

RubenKelevra, 2016-05-09:
Bitte wieder öffnen, das Problem ist nicht gefixt.

NeoRaider, 2016-05-09:
Es ist für die zukünftige Entwicklung von Gluon geplant, die Trennung zwischen Mesh und WAN durch Network Namespaces noch zu verstärkten, dadurch soll es Prozessen auf dem Knoten auf Dauer tatsächlich unmöglich werden, den WAN-Port überhaupt zu sehen.

Sorgt dafür, dass euer Mesh funktioniert, dann habt ihr kein Problem.

RubenKelevra, 2016-05-09:
> Sorgt dafür, dass euer Mesh funktioniert, dann habt ihr kein Problem.

Was soll deine Provokation?

Dieses Problem ist für alle die im Internet Update-Server betreiben ein Problem und ich sehe grade nicht warum du dieses Ticket ohne Diskussion ablehnst.

Insbesondere bei dem bei vielen anstehenden inkompatiblen Update, wäre hier eine Dokumentation dieses Verhaltens wohl dringend notwendig.

Ich verstehe den Hintergrund des Sicherheitsaspekts, nur ist der Update-Dienst wohl einer der Dienste die von Haus aus so sicher sein sollten, dass sie wohl über den Uplink DNS machen dürfen, oder nicht? Immerhin geht hier sonst die Möglichkeit kaputt überhaupt Updates zu fahren, wenn zum Beispiel der VPN-Dienst streikt.

Wir haben eine Firmware gebaut und ausgerollt, die für einzelne Geräte defekt zu sein scheint. Die selbe Firmware in der selben Build-Umgebung erneut gebaut funktioniert. Nun möchten wir diese ausrollen und durch dieses nicht-dokumentierte Verhalten müssen wir vor Ort die Geräte updaten, per Hand.

Zusätzlich dazu, ließt sich eure Dokumentation an der Stelle nicht so als wäre es vollkommen unsinnig ins Internet Update-Server zu stellen:

`autoupdater = {`
`  branch = 'stable',`
`  branches = {`
`    stable = {`
`      name = 'stable',`
`      mirrors = {`
`        'http://[fdca:ffee:babe:1::fec1]/firmware/stable/sysupgrade/',`
`        'http://autoupdate.entenhausen.freifunk.net/firmware/stable/sysupgrade/',`
`      },`
`      -- Number of good signatures required`
`      good_signatures = 2,`
`      pubkeys = {`
`        'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX', -- someguy`
`        'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX', -- someother`
`      }`
`    }`
`  }`
`}`
`

NeoRaider, 2016-05-09:
Entschuldigung, eine Provokation war nicht meine Absicht.

Ich muss zugeben, dass unsere Dokumentation in der Hinsicht unklar ist, ich werde mich darum kümmern, das klarzustellen.

Aus Interesse: Welche Art von Update macht ihr gerade, die zu diesen Problemen führt? Ich denke, die meisten Communities arbeiten bei solchen Updates so, dass ein Teil der VPN-Server aufs neue Setup umgestellt wird (mit neuem Port, sodass sich alte Knoten nicht mit neuen Servern verbinden und umgekehrt), und so sichergestellt ist, dass keine Knoten aus dem VPN fallen, bis alle Knoten aktualisiert sind.

jplitza, 2016-05-11:
@RubenKelevra Da eine Verbindung zum Mesh fast sicher auch eine Verbindung ins Internet bedeutet, kann man Update-Server auch "ins Internet" stellen, ja.

Tatsächlich handelt es sich vermutlich um einen Bug in die andere Richtung: Wenn ein Update-Server eine öffentliche IPv4-Adresse hat, benutzt der Knoten beim Autoupdate das WAN-Interface. Das sollte glaube ich nicht so sein (sowohl unter dem Gesichtspunkt "das Autoupdate sollte *immer* übers Mesh laufen", als auch weil ein Knoten ohne WAN-Anschluss das nicht kann). (Und falls der Update-Server eine interne IPv4-Adresse hat versucht der Knoten sie von der local-node-Adresse zu erreichen, was natürlich auch nicht funktioniert…)

Zum Sicherheitsaspekt: Es geht nicht darum, dass der Dienst sicher genug ist um ihn über den WAN-Anschluss zu betreiben. Das unverschlüsselte Mesh ist bestimmt nicht sicherer. Es geht darum, dass man einem Netzwerkadministrator beispielsweise in einer Behörde sagen kann "Der Knoten baut nach einem DNS-Lookup eine UDP-Verbindung zu diesem Port auf, alles andere könnt ihr blocken".

Adorfer, 2016-05-11:
Ich sehe in der Argumentation einige vermutlich nicht zulässige Umkehrschlüsse. 
Wenn es einen präferierten Weg gib: Sollte man dann andere Möglichkeiten explizit unterbinden? Oder sollte man es trotzdem ermöglichen und ggf. ein paar Stolpersteiner (hier: DNS) aus dem Weg räumen?

Wenn man eine public-FQDN(zu den er nur A-Records gibt) oder PublicIPv4 als Updateserver in der site.conf angibt, dann ist das für mich eine Entscheidung der Community (der Firmwarebauenden), dass die images ggf. auch direkt "via WAN" geholt werden sollen 

Man kann sagen, dass es Szenarien gibt, in denen das nicht gewüncht wirde. 
Rein rechtlich sehe ich aber kein Problem darin, wenn ein Knoten, der versucht eine PublicIPv4 via WAN zu erreichen für ein wget eines syupgrades, zusätzlich auch noch vorher einen DNS-Query für die FQDN absetzt. 

**Aber sicherheitshalber nochmal betont: "Schön" ist anders.** 
Aber wenn man schon der Wget ggf. mit dem source des br-wan rausgeht, dann schadet der dns-lookup direkt davor auch nimmer.