freifunk-gluon/gluon #762: /etc/rc.local wird nicht ausgeführt beim Start

Adorfer, 2016-05-12:
Perhaps it's an upstream issue: 

in CC it looks like "poe-secondary"-scripts can not be ran i directly any more like in BB. 
perhaps it's "too early". 

     echo 20 > /sys/class/gpio/export
     echo out > /sys/class/gpio/gpio20/direction
     echo 1 > /sys/class/gpio/gpio20/value

does not work on CPE210 any more. 
(with a subshell and a 30 second delay in front: works, but i am not happy with that solution.)

I am not sure if this a new issue or could be linked to 
#328 
But possibly other actions are affected too, that some stuff is not populated at the time the rc.local is called at the moment.

NeoRaider, 2016-05-12:
This is a consequence of #651 (and the corresponding upstream backport). We now have proper UCI-based PoE-passthrough support, so the setting made in rc.local is overwritten by the setting from */etc/config/system*... The update to the current OpenWrt CC HEAD has brought this change to Gluon v2016.1.4 as well.

It's a bit weird that the GPIO setup script runs after rc.local, but I guess we can live with it...

closed 2016-05-12 by NeoRaider

Adorfer, 2016-05-12:
#328 is open, so what is the propper UCI-statement to use PPOE-chaining?

NeoRaider, 2016-05-12:
Just look what it is called in /etc/config/system...

It should be something like `uci set system.gpio_switch_poe_passthrough.value=1`.