freifunk-gluon/gluon #765: vi hängt

4ndr3, 2016-05-13:
Dateien lassen sich reproduzierbar nicht editieren -> ssh-Sitzung bleibt hängen.
Beobachtet auf 
- Ubiquiti Nanostation M2 mit $ vi /etc/config/wireless
- TP-Link TL-WDR4900 mit  $ vi /etc/config/network

Kein Eintrag in logread. Dateien lassen sich mit nano editieren. Andere Dateien lassen sich mit vi editieren.
Firmware bei beiden Geräten: Hamburg 0.8 / gluon-v2016.1.4

Adorfer, 2016-05-13:
und das ist jetzt nicht nur ein batman15-Refragmentation-Bug, weil auf einem Supernode ein ungepatchter Batman läuft? 

Setzte mal testweise auf dem ssh-"quellrechner" die MTU des br-client-Interfaces aus 1280. Nur zum Test bitte.

ohrensessel, 2016-05-15:
Hamburg nutzt batman-adv 2013.4

NeoRaider, 2016-05-26:
Da ich das Problem hier noch nie gesehen habe, wäre es sehr hilfreich, wenn ihr mit tcpdump weitere Informationen sammeln könntet.

kantorkel, 2016-05-30:
Mich interessiert die Ausgabe von `df -h`. Ähnliches Verhalten kenne ich bisher nur von Geräten, auf denen kein Speicherplatz oder keine inodes mehr frei waren.

NeoRaider, 2016-07-27:
Closing because of inactivity. Please reopen if you can provide more information about this.

closed 2016-07-27 by NeoRaider