freifunk-gluon/gluon #766: respondd send's responds over ipv4 default gateway

lichtfeind, 2016-05-15:
respondd answers packages send from a spoofed ipv4 address to the local/next_node ipv4 address of the access point (10.38.0.127 for Braunschweig) by sending the answer over the wan interface.
The spoofed  IPv4 is from the network connected to the wan port or any other ip from the internet

Tested on a TL-WR841N gluon v2016.1.2

NeoRaider, 2016-05-15:
Thanks for the report. We added firewall rules restricting respondd to IPv6 link-local addresses some time ago, but I was not aware that respondd was responding to IPv4 queries at all, so the rules don't cover that yet. I'll fix this as soon as possible.

NeoRaider, 2016-05-16:
Fixed in freifunk-gluon/packages@63376e23c81e53c21d3c5250c3fb7444a90dc019, v2016.1.x has been updated as well.

closed 2016-05-16 by NeoRaider

jplitza, 2016-05-16:
Just to assess the security threat posed by this issue: You still have to be connected to the node your are trying to provoke an answer from, as the IP and even MAC address of the next-node interface is not normally routed in the mesh.

lichtfeind, 2016-05-16:
I am not sure what would happen if you use a layer2 broadcast address. That might not be possible over a normal node with would route the package to the local-node "interface", but one could setup an "evil" node which let's this packing in layer 2 batman network loss.
At the moment i don't have a setup were i can safely test this.

lichtfeind, 2016-05-17:
I tested the broadcast theory and it works for an "evil" node but not over a regular gluon node.