freifunk-gluon/gluon #770: L2-Separation of Concerns, Act 2

T-X, 2016-05-18:
Observations after #688:

* Multicast noise is greatly reduced on the local network segment of a node.
* Some rare unsolicited Neighbor Solicitations to ff02::1 remain (usually with an override flag set, so a node changing it's address?)
* Most remaining traffic is ARP, and these consist of:
  * Mostly ARP requests from the gateways to determine the MAC address of a client (ARP requests from clients seem to be handled fine by DAT)
  * Also a few [ARP probes](https://en.wikipedia.org/wiki/Address_Resolution_Protocol#ARP_probe)
* Multicast snooping happens after bridge fdb (forwarding database) updates, unfortunately (but netfilter PREROUTING hooks come before and INPUT/FORWARD after these two things)
* For Hamburg still a few hundred entries remained in the bridge fdb due to the reasons above, which are a little redundant to the batman-adv translation table.

ToDos / Ideas:

* Ask Antonio, why DAT might not work ideal for the GW->Client ARP request direction
  * A GW must have received a prior ARP request from Client->GW and even an IP packet from Client->GW: Is there an option in the kernel IP neighbor table to learn from these, too (or at least use them as a hint for a unicast ARP request)? Or can DAT spoof these packets, too (or does it already, but not ideally?)? Or is this ARP request a result of the DHCP server performing a duplicate address detection?
* Similar to as it's done for multicast by forwarding everything towards bat0 (bridge multicast_router option enabled for bat0), also set brport/learning=off on bat0 and brport/unicast_flood=on. This would disable the bridge fdb for bat0 (learning=off). Looking at the bridge code (need to verify with tests), unicast_flood will always forward to bat0 if (and only if) no fdb entry is present for the destination MAC, instead of flooding to all ports.
* Filter ARP probes? It's a relatively new standard, so isn't mandatory. Also, gateways should perform duplicate address detection on their own before handing out IP addresses already.
* Filter unsol. IPv6 Neighbor Solicitations to ff02::1? Any protocol using ff02::1 doesn't really work in such large l2 mesh netwoks as is. Make this an unsupported feature for now by filtering such messages? (should only be needed for devices changing their address / hotplugging with high availability)

(Act 3: Patchset for multicast optimizations in batman-adv for bridged setups finally got merged, will come with one of the next release -> multicast overhead reduction on VPN and adhoc/802.11s interfaces)

mweinelt, 2018-11-25:
@T-X Can you give an update on the state of this issue?

T-X, 2019-01-11:
* Disabling the bridge FDB learning was postponed for now. The first try had some issues and needed to be reverted. The memory consumption reduction is probably marginal anyways right now, though.

Regarding traffic:

* ARP:
  * batman-adv DHCPACK snooping: The [according patch](https://patchwork.open-mesh.org/patch/17721/) has gotten some initial reviews and is in its 8th iteration now. Will hopefully be applied upstream soon.
  * batman-adv extended DAT cache timeout: I've submitted a second variant right now [here](https://patchwork.open-mesh.org/patch/17728/)
  * Both together hopefully significantly reduce the number of ARP Requests we see from gateways

* ICMPv6 Neighbor Solicitations:
  * A multicast-to-multi-unicast patch for batman-adv is provided in #1357 and there were some positive test results. However, before it can be accepted upstream the [batman-adv sysfs->netlink changes](https://patchwork.open-mesh.org/project/b.a.t.m.a.n./list/?series=188&state=%2A&archive=both) need to go in first as otherwise the Linux network subsystem maintainers will likely reject any patch which introduces a new configuration option.

T-X, 2019-01-11:
Also:

The batman-adv multicast announcement and storage part is active since Gluon 2018.1. Several more bugs in the batman-adv translation table were fixed last year (special thanks to @lemoer). You will now notice multicast MAC addresses in "batctl tl" and "batctl tg" for instance. In real setups you will probably not see a difference in traffic with this "multicast optimizations" feature yet. As a single not-yet-updated node or an inconveniently placed snooping switch will render it void.

However #1357 is supposed to change that. In a mesh network which has mostly updated to Gluon >=2018.1 you can already install and try #1357 on a few nodes without needing to update the whole network with this pull-request. #1357 also allows you to play with multicast streaming from such nodes already.

T-X, 2019-01-11:
Finally, another layer 2 concern was tackled with the **gluon-ebtables-limit-arp** [package](https://gluon.readthedocs.io/en/v2018.2/package/gluon-ebtables-limit-arp.html#).

More and more "smart" phones appeared which seemingly had some genius app trying to scan the whole subnet. Which introduced a lot of ARP traffic and overall burdon in a mesh network. gluon-ebtables-limit-arp rate limits such requests.

It was introduced with Gluon 2018.1 (84a6f65f02d7e36a073ba2839712f0c0bb1dda10) and became opt-out (installed by default) in Gluon 2018.2 (1a690d2fd6fc564532d266035ba114940f4393b5).

neocturne, 2025-04-04:
@T-X Are there still any points open from this issue, and if so, should they be moved to individual tickets?

T-X, 2025-04-04:
I think this is mostly done.

For the first DAT ToDo, it's still WiP here but I hope it gets merged soon, so no need to track here further: https://patchwork.open-mesh.org/project/b.a.t.m.a.n./cover/20250117021348.2021-1-linus.luessing@c0d3.blue/

For ARP probes and unsol. IPv6 Neighbor Advertisements to ff02::1, I don't see them that often anymore, so negligible, no need/importance to track here.

Disabling bridge unicast/FDB learning on bat0 was an idea (and unsuccessfully tried and reverted once: 1ed87c6df401e63) when we were low an RAM, but not important anymore as we dropped those low RAM targets.

So this ticket can be closed, I think.

closed 2025-04-04 by T-X