freifunk-gluon/gluon #772: fastd-key does not get displayed

kpanic23, 2016-05-19:
Since gluon 2016.1.4 we have noticed following quirk:
You flash the factory image. After configuring the node with VPN mesh and clicking "save", the router reboots. Normally the VPN key should get displayed. Instead the browser times out. Somehow the reboot seems to happen too early, rendering the browser unable to load the page.
This seems to happen only on a newly flashed device. If config mode is entered via reset button, everything works ok.

NeoRaider, 2016-05-19:
Interesting... how reproducible is this, and on what hardware are you seeing this behaviour?

I've tried it on a TL-WR1043ND v1 with the current v2016.1.x and can't reproduce the issue.

kpanic23, 2016-05-19:
I have noticed it on a TL-WR841N v9 and a TL-WR941ND v6.
A co-Freifunker had the same issue with a TL-WR940N v3

Firmware: http://ftp.ff3l.net/netsplit/output/hoho/stable/factory/

NeoRaider, 2016-05-19:
Okay, I can reproduce the issue using your firmware. I'll look into it.

NeoRaider, 2016-05-19:
Your firmware does not contain haveged. Including haveged is necessary, as on most embedded systems, the kernel doesn't generate enough entropy by default, so generating the fastd secret will take a long time.

When I comment out the 'reboot' call in your firmware, the page will just hang loading and eventually timeout, because the key generation will take several minutes.

Still, the reboot should wait for the key to be generated in any case, so there's still a (minor) bug to fix.

kpanic23, 2016-05-19:
Mea culpa,
I have just noticed (ok, been noticed...) that by commenting out a package in our site.mk, all following packages also get commented out. This unfortunatelly includes haveged.

mmalte, 2016-05-19:
actually all the the GLUON_SITE_PACKAGES entries in the site.mk are in one line, the \ prevents a word-wrap

commit 97879e80 (2016-05-19): gluon-config-mode-core: clean up reboot handling

Don't fork reboot process before all package hooks have been handled and
rendering is complete.

Replace debug.setfenv hack to close stdout with nixio.dup.

Fixes #772
closed 2016-05-19 by NeoRaider