freifunk-gluon/gluon #776: build fails for "node": " error: looser throw specifier for 'virtual node::StreamBase::~StreamBase()'"

T-X, 2016-05-21:
With a current gluon master I'm unable to build ar71xx, even after a "make dirclean". The following error is reported on this Debian system:

<pre>
[...]
make   -C /home/tux/dev/gluon3/build/ar71xx-generic/openwrt/build_dir/host/node-v4.4.4  
make[5]: Entering directory `/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4'
make -C out BUILDTYPE=Release V=s
make[6]: Entering directory `/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out'
  g++ -pthread -rdynamic -m64  -o /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/cctest -Wl,--start-group /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/cctest/test/cctest/util.o /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/deps/gtest/libgtest.a /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/deps/v8/tools/gyp/libv8_libplatform.a /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/deps/v8/tools/gyp/libv8_base.a /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/deps/v8/tools/gyp/libv8_libbase.a /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/deps/v8/tools/gyp/libv8_nosnapshot.a -Wl,--end-group -ldl -lrt
  g++ '-DNODE_ARCH="x64"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DHAVE_OPENSSL=1' '-D__POSIX__' '-DHTTP_PARSER_STRICT=0' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_POSIX_C_SOURCE=200112' -I../src -I../tools/msvs/genfiles -I../deps/uv/src/ares -I/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj/gen -I../deps/v8 -I../deps/cares/include -I../deps/v8/include -I../deps/openssl/openssl/include -I../deps/zlib -I../deps/http_parser -I../deps/uv/include  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -ffunction-sections -fdata-sections -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++0x -MMD -MF /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/.deps//home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/node/src/js_stream.o.d.raw   -c -o /home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/node/src/js_stream.o ../src/js_stream.cc
In file included from ../src/js_stream.h:6:0,
                 from ../src/js_stream.cc:1:
../src/stream_base.h:225:11: error: looser throw specifier for 'virtual node::StreamBase::~StreamBase()'
../src/stream_base.h:141:11: error:   overriding 'virtual node::StreamResource::~StreamResource() noexcept (true)'
In file included from ../src/js_stream.cc:1:0:
../src/js_stream.h:17:3: error: looser throw specifier for 'virtual node::JSStream::~JSStream()'
In file included from ../src/js_stream.h:6:0,
                 from ../src/js_stream.cc:1:
../src/stream_base.h:225:11: error:   overriding 'virtual node::StreamBase::~StreamBase() noexcept (true)'
make[6]: *** [/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out/Release/obj.target/node/src/js_stream.o] Error 1
make[6]: Leaving directory `/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4/out'
make[5]: *** [node] Error 2
make[5]: Leaving directory `/home/tux/dev/gluon3/openwrt/build_dir/host/node-v4.4.4'
make[4]: *** [/home/tux/dev/gluon3/build/ar71xx-generic/openwrt/build_dir/host/node-v4.4.4/.built] Error 2
make[4]: Leaving directory `/home/tux/dev/gluon3/packages/openwrt/lang/node'
make[3]: *** [package/feeds/module_openwrt/lang/node/host/compile] Error 2
make[3]: Leaving directory `/home/tux/dev/gluon3/build/ar71xx-generic/openwrt'
make[2]: *** [/home/tux/dev/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.package_compile] Error 2
make[2]: Leaving directory `/home/tux/dev/gluon3/build/ar71xx-generic/openwrt'
make[1]: *** [prepare] Error 2
make[1]: Leaving directory `/home/tux/dev/gluon3/build/ar71xx-generic/openwrt'
make: *** [all] Error 2
make GLUON_TARGET=ar71xx-generic V=s  83,22s user 20,68s system 33% cpu 5:09,33 total
</pre>

T-X, 2016-05-22:
Issue is on a Debian Wheezy, so Debian oldstable. A Debian Sid/unstable builds fine. Therefore closing it again, as I suppose Debian oldstable isn't supposed to be supported by OpenWRT.

closed 2016-05-22 by T-X

NeoRaider, 2016-05-22:
I agree. While the OpenWrt core should stay buildable on Debian Wheezy, I don't think we can make that promise for host builds of all community packages. For Gluon, this makes Debian Jessie the minimum requirement, at least if gluon-status-page is included.

waschmi, 2016-05-22:
That makes me sad. I thought I still had a bit of time (wheezy end of life) before the systemd madness affect the use of my computer. With jessie I will have to watch out which packages will pull that dependency. I understand Gluon can't support 'ancient' things forever but my (computer) life just now got an order of magnitude harder.

rotanid, 2016-05-22:
@waschmi have a look at devuan.
but you could also "simply" blacklist systemd, search for "How to stay with sysvinit in Debian Jessie"

NeoRaider, 2016-05-28:
I've decided to downgrade nodejs again and backport the GCC 6 patch. This should allow building on Wheezy as long as we're on OpenWrt CC.