freifunk-gluon/gluon #778: Multicast: dont block local peer discovery for syncthing

lemoer, 2016-05-21:
Syncthing is kind of an opensource alternative to BitTorrentSync. I think it would be very nice, if we could add ebtables rules to allow the [local peer discovery](https://docs.syncthing.net/specs/localdisco-v3.html).

Adorfer, 2016-05-24:
please make it at least configurable in the site.conf plus disable it by default. 

(sorry, but any broadcast traffic on the network can break the neck of an already "big" network. )

lemoer, 2016-05-24:
Local peer discovery is about 1 packet per 30-60 seconds. If this will break your network then your network will also break without this "traffic" soon. But local peer discovery brings another nice aspect, since the exchanged payload data (exchanged via unicast) won't go through any exit node and so it doesn't bind more resources than it really has to.

Please keep in mind, that we try to build a cool and decentral network and not a highly optimized commercial internet access solution ;)

NeoRaider, 2016-05-26:
We already allow BitTorrent local peer discovery, so adding Syncthing support would make sense. The gain from the reduced traffic between the mesh and the internet should more than outweigh the small increase in broadcast traffic if anybody actually uses Syncthing (and if almost nobody uses it, the additional broadcast traffic will be negligible anyways).

One thing to note for all of these local peer discovery protocols: They won't work anymore as soon as we switch to a layer-3-mesh.

In the long run, it would be great if we could find a way to fit a Avahi proxy into our firmwares (one that is made for mesh networks and doesn't generate more traffic than necessary) - then all such softwares like Syncthing would just need to implement Avahi/Bonjour instead of inventing the wheel again and again ;)

NeoRaider, 2016-05-26:
Making this configurable in the site.conf would of course be a very nice addition as well. Doing so should be simple, as the ebtables rule files already use a Lua-based DSL.

christf, 2017-01-07:
I feel that it makes sense to differentiate node-local, site-local an network-wide announcements. The current scheme that is utilizing ff02 and ff05 does not support this. 

Mmfd could play a  role when forwarding multicast traffic in a l3roamd mesh.

NeoRaider, 2020-05-29:
Closing in favour of the more generic issue #2020.

closed 2020-05-29 by NeoRaider