freifunk-gluon/gluon #784: *** No rule to make target 'package/feeds/module_openwrt/lang/lua-cjson/host/compile', needed by 'package/feeds/gluon/gluon-site/compile'.  Stop

T-X, 2016-05-23:
Not quite sure how I managed to get into this state. I think I had somehow, accidentally executed some "make clean" or "make dirclean" command within the checked out openwrt repository and aborted it after a few seconds.

Running "make dirclean" in gluon did not seem to fix it. The log at the end with "make GLUON_TARGET=ar71xx-generic V=s" looks like this:

<pre>
make[4]: Leaving directory '/mnt2/gluon3/openwrt/package/system/fstools'
make[4]: Entering directory '/mnt2/gluon3/openwrt/package/boot/uboot-envtools'
. /mnt2/gluon3/build/ar71xx-generic/openwrt/include/shell.sh; bzcat /mnt2/gluon3/build/ar71xx-generic/openwrt/dl/u-boot-2014.10.tar.bz2 | tar -C /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.

Applying ./patches/001-compile.patch using plaintext:
patching file tools/env/Makefile

Applying ./patches/100-fw_env_gnu_source.patch using plaintext:
patching file tools/env/fw_env.c

Applying ./patches/200-fw_env_no_aes.patch using plaintext:
patching file tools/env/fw_env.c

Applying ./patches/300-support-env-in-ubivol-chardev.patch using plaintext:
patching file tools/env/Makefile
patching file tools/env/fw_env.c
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/.prepared_cabc81b44c079e6dee66af11e235a80c
rm -f /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.uboot-envtools_installed
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/include/config.mk
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/include/config.h
rm -f /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/.configured_*
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/.configured_nnnnnnnn
make -C /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10 CROSS_COMPILE="mips-openwrt-linux-uclibc-" TARGET_CFLAGS="-Os -pipe -mno-branch-likely -mips32v
make[5]: Entering directory '/mnt2/gluon3/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
/mnt2/gluon3/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[7]: *** [silentoldconfig] Error 1
/mnt2/gluon3/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/Makefile:464: recipe for target 'silentoldconfig' failed
make[6]: *** [silentoldconfig] Error 2
  HOSTCC  tools/env/aes.o
  HOSTCC  tools/env/crc32.o
  HOSTCC  tools/env/ctype.o
  HOSTCC  tools/env/env_attr.o
  HOSTCC  tools/env/env_flags.o
  HOSTCC  tools/env/fw_env.o
  HOSTCC  tools/env/fw_env_main.o
  HOSTCC  tools/env/linux_string.o
  HOSTLD  tools/env/fw_printenv
  STRIP   tools/env/fw_printenv
make[5]: Leaving directory '/mnt2/gluon3/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10'
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/.built
mkdir -p /mnt2/gluon3/build/ar71xx-generic/openwrt/bin/ar71xx/packages /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/CONTo
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/usr/sbin
install -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/tools/env/fw_printenv /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_3n
ln -sf fw_printenv /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/usr/sbin/fw_setenv
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/lib
install -m0644 ./files/uboot-envtools.sh /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/lib
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/etc/uci-defaults
install -m0644 ./files/ar71xx /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/etc/uci-defaults/30_uboot-envtools
find /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm f
export CROSS="mips-openwrt-linux-uclibc-" NO_RENAME=1 ; NM="mips-openwrt-linux-uclibc-nm" STRIP="/mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/host/bin/sstrip" STRIP_KMOD="/mnt2/gluon3/build/ar71xx-generic/openwrt/ss
rstrip.sh: /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/ipkg-ar71xx/uboot-envtools/usr/sbin/fw_printenv: executable
[... 9 lines cut ...]
mkdir -p /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/stamp /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gs
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/usr/sbin
install -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/u-boot-2014.10/tools/env/fw_printenv /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mipsn
ln -sf fw_printenv /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/usr/sbin/fw_setenv
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/lib
install -m0644 ./files/uboot-envtools.sh /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/lib
install -d -m0755 /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/etc/uci-defaults
install -m0644 ./files/ar71xx /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools/etc/uci-defaults/30_uboot-envtools
SHELL= flock /mnt2/gluon3/build/ar71xx-generic/openwrt/tmp/.root-copy.flock -c 'cp -fpR /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envto'
rm -rf /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/tmp-uboot-envtools
touch /mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/root-ar71xx/stamp/.uboot-envtools_installed
make[4]: Leaving directory '/mnt2/gluon3/openwrt/package/boot/uboot-envtools'
make[4]: Entering directory '/mnt2/gluon3/packages/luci/modules/luci-base'
make[4]: Leaving directory '/mnt2/gluon3/packages/luci/modules/luci-base'
[... cut ...]
package/Makefile:188: recipe for target '/mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.package_compile' failed
make[2]: *** [/mnt2/gluon3/build/ar71xx-generic/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/stamp/.package_compile] Error 2
make[2]: Leaving directory '/mnt2/gluon3/build/ar71xx-generic/openwrt'
/mnt2/gluon3/Makefile:341: recipe for target 'prepare' failed
make[1]: *** [prepare] Error 2
make[1]: Leaving directory '/mnt2/gluon3/build/ar71xx-generic/openwrt'
Makefile:71: recipe for target 'all' failed
make: *** [all] Error 2
</pre>

T-X, 2016-05-23:
Removing the openwrt directory completely and performing a clean checkout via "make update" helped.

closed 2016-05-23 by T-X

NeoRaider, 2016-05-23:
Yes, executing *any* make command within the *openwrt* directory will usually break the tree. We could probably patch away the toplevel makefile there (I don't think we use it) to avoid such mistakes.