freifunk-gluon/gluon #787: Add option to add second ssid for 5GHz

mmalte, 2016-05-26:
It would be great to add a site.conf option to add a second ssid to the 5GHz Band.

That Users get the possibility to force their device to 5GHz while still everybody who do not care might get connected if his device is clever enough.

We are running this setup in our dormitory network and the number of users in the 5GHz band have increased significantly.

The only con is that some airtime is wasted, but that is not such a big issue in 5GHz.

RubenKelevra, 2016-07-26:
Actually this is technically not the right solution. A SSID is a network-name, since both wifis are the same network, both should be named equally.

The devices seem to be very clever at the moment, we got 1/3 of the clients of 2,4 GHz on the 5 GHz band es well on indoor locations. 

I think we should just increase the 5 GHz Wifi and reduce the 2,4 ones. Especially on wide range applications like bus stations or train stations. :)

NeoRaider, 2016-07-27:
This is a duplicate of #474.

We're also working on some roaming helper at the moment, which might also be used for band steering in the future.

closed 2016-07-27 by NeoRaider