freifunk-gluon/gluon #789: include new packages: hoods/ hoodselector

FreifunkUFO, 2016-05-28:
Hookselector sounds nice!
there should be (3) packages inside this repo
https://blog.nordwest.freifunk.net/2016/05/20/neue-firmware-0-9-hoodselector-und-support-fuer-wr841ndv11/

code atm near:
https://git.nordwest.freifunk.net/ffnw-firmware/packages/blob/a1b06b6646c4b51129cb16575776b65f5b278b15/hoodselector/luasrc/hoodselector

kb-light, 2016-05-29:
Yes it is a cool feature (we also provide something similar, see below), but why do you need this upstream? Cann't you just use the repo of Freifunk Nordwest, if you want to include that package into your firmware?

Additionally there are some similar implementations:

- site-select by DO9XE: https://github.com/freifunk-gcp/v2015.2
- our (FFHO) site-select-Packages: https://git.c3pb.de/freifunk-pb/ffho-packages/tree/v2016.1.x/ffho
- ...

NeoRaider, 2016-05-29:
Well, if multiple communities are interested in the same feature, getting this upstream makes sense. At the moment, too many nice features are hidden in community repos...

FreifunkUFO, 2016-05-29:
please post a screenshot of the wizard with hoodselecting capability.
i cant see it with ffnw firmware (installed on a router) or on https://hochstift.freifunk.net/anleitung-router-flashen-und-konfigurieren/

mmalte, 2016-05-31:
We are doing something similar without adding any packages to gluon:

https://github.com/ffac/segmenter

Freifunk Stuttgart does it quite similar. 

There are so many ways to archive this out there, I don't think it's a good idea to include some random solutions.

PowerPan, 2016-06-01:
Everyone who don't know what the Hoodselector is, here is a actual video from @2tata https://www.youtube.com/watch?v=rlRTGez7LEI

2tata, 2016-06-01:
Hi,

> Everyone who don't know what the Hoodselector is, here is a actual video from @2tata https://www.youtube.com/watch?v=rlRTGez7LEI

Next month I would like to enhance the code quality and port the hoodselector to C++. After that I can send a patch or create a pull request. Currently I am nearly before my exsams, so I think It is ok for all if the upstream patch will just come in a few month. :)

Also we on the Nordwest Freifunk community need more time to test the hoodselector. There are currently some Issues open. 
https://git.nordwest.freifunk.net/ffnw-firmware/packages/issues

vg
Tarek

2tata, 2016-06-01:
the hoodselector is a half automated and decentralize layer 2 network segmentation tool without a manual selection from users on the config mode. The hoodselector chose a router hood depending on geo-coordinates and neighbour Routers. We have an issue for manual selecting of hood but that is actual a discussion point because technically speaking the hoodselector does not is the right way for manual cousins of hoods.

kb-light, 2016-06-02:
For manually selection we have the [ffho-config-mode-site-select](https://firmware.hochstift.freifunk.net/config-mode-neu.png) package

2tata, 2016-06-02:
The hoodselector does not work with the siteconf

Adorfer, 2016-06-02:
@NeoRaider 
 >  At the moment, too many nice features are hidden in community repos...

Slightly offtopic: Compare this to the discussion on the ML about DE-de/qwertz-layout for x86-images.
A lot of people not make a 2nd proposal/pr after such a debate.

FreifunkUFO, 2016-08-05:
Any news about that project??

Is hoodselector only allowed to be used on "Freifunk Nördwest"? would be nice to have it in other communities / firmwares!  also would be nice to have a manual selection from user (at least for communities which wants that)

mweinelt, 2016-08-05:
Well, someone would have to step up and make a pull request. As the hoodselector is rather complex there should also be documentation and integration into site.conf, so that might be some work.

2tata, 2016-08-09:
Hi,

> Well, someone would have to step up and make a pull request. As the hoodselector is rather complex there should also be documentation and integration into site.conf, so that might be some work.

Currently I spend most of my time to my Google Summer of code Project. I did some bugfixing and code quality enhancement for the hoodselector but there still one major issue before I will send a patch to gluon. At the moment there no mesh on lan/wan handling. After my GSoC project I plan to implement this. but I need more time.

cheers
Tarek

FreifunkUFO, 2016-12-30:
nice, now there is a pull request for gluon firmware:
https://github.com/freifunk-gluon/gluon/pull/997

2tata, 2016-12-30:
On 12/30/16 11:19, FreifunkUFO wrote:
> nice, now there is a pull request for gluon firmware:
> https://github.com/freifunk-gluon/gluon/pull/997

right, I'll update this PR soon because there are now many
changes between the patches that I send some weeks before
and the current version in our repo.

cheers
Tarek

rotanid, 2018-01-19:
#1226 is the place for discussions, closing this.

closed 2018-01-19 by rotanid