freifunk-gluon/gluon #790: Packages with Openembedded

MTRNord, 2016-05-29:
Can I build with Openembedded and meta-openwrt (bitbake plugin) build packages for gluon? Or is there a way how can I get it to work?

NeoRaider, 2016-05-29:
As far as I can tell, meta-openwrt doesn't provide a bitbake plugin, but is a layer that contains many of OpenWrt's core packages (libubox, ubus, procd, ...) to be used in an OpenEmbedded-based distribution. I don't think it is usable to create OpenWrt packages.

It could be a starting point to port Gluon over to OpenEmbedded, but such a step will definitely take a lot of effort.

MTRNord, 2016-05-29:
Ok than I misinterpreted the package I think. :/ my main goal was to get meta-nodejs compiled for gluon because it is much more leightweight than nodejs itself

MTRNord, 2016-05-29:
I have found an easier way to Port Nodejs but is there a var for Makefiles of Packages to get the toolchain of the router image that is build when running the makefile?

closed 2016-05-30 by MTRNord