freifunk-gluon/gluon #796: Receiver-Bug in CPE210/CPE510

Sunz3r, 2016-06-03:
Both devices needs to enable the Low Noise Amplifier via GPIO:

```
echo 18 > /sys/class/gpio/export
echo 19 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio18/direction
echo out > /sys/class/gpio/gpio19/direction
echo 1 > /sys/class/gpio/gpio18/value
echo 1 > /sys/class/gpio/gpio19/value
```
http://dev.ffggrz.de/sunzi/cpe210_lna.png

This is already patched in LEDE:
http://lists.infradead.org/pipermail/lede-dev/2016-May/000588.html

https://dev.openwrt.org/ticket/19402#comment:16

Adorfer, 2016-06-04:
i do down know where it helps, but my attempts on 2 CPE210v1.1 and 5 CPE210v1.0 showed *loss* of abount 20-40% TQ after sending those commands to the GPIOs.

viisauksena, 2016-06-06:
ich kann für cpe210v1 bestätigen das der empfang (=reichweite) enorm besser geworden ist ... 
die drei cpe in dem dach (in drei versch. richtungen , an drei versch stellen im dach) hatten vorher kaum connections ...
http://openfreiburg.de/freifunk/meshviewer/#!v:m;n:60e327531f9e
- wenn gewünscht kann ich auch bestimmte(!) genauere Dinge testen

plumpudding, 2016-06-14:
It doesn't seem to work:
https://map.ffgl.eu/#!v:m;n:60e327d6125c
These connections were much better before.

kpanic23, 2016-06-14:
I've tried it.
Signal strength increased, TQ went from ~45% to ~95%.
I even liked it so much, I applied the patch to 2016.1.5 and rolled out the firmware via autoupdate for all our CPEs.
No problems so far, just better mesh links!

Here is a screenshot of the status page of my CPE210 1.0 right after executing the commands from OP:
[image]

NeoRaider, 2016-06-14:
@plumpudding, are you actually using the current master with the kernel patch (with deac14e62959f9cc3aba1d974950c092224e78d0), or the replacement commands posted by @Sunz3r in this issue? Just enabling these GPIOs is probably wrong, you need to set these GPIOs to LNA mode in the kernel.

plumpudding, 2016-06-14:
I use the latest gluon master.

plumpudding, 2016-06-16:
I tried the /proc variant too. In both cases the dBm get better, while the tq becomes unstable and alot worse.

NeoRaider, 2016-06-17:
I've reverted the change for now, because of the TQ issues, and https://git.lede-project.org/?p=source.git;a=commit;h=c2e0c41842895ba47819fa98b785c76a2524628b ...

mokkin, 2016-06-23:
For my CPE510 v1.1 the patch works very well. I have a connection about 450m between two CPE510. Before patching the connection was very weak and got lost sometimes. TQ around 45%. Because I don't have permanente access to the second CPE I have patched only my one until now and the signal raised to TQ 100%, -71dBm.
[image]

Gammelknup, 2016-06-23:
maybe the root cause of  problem, which some of you has seen is caused by the switching the LAN and WAN port? Can some of you retest:
https://github.com/freifunk-gluon/gluon/issues/806

mokkin, 2016-07-10:
Today I applied the patch also on the other CPE510 v1.0 and the dBm has become even better:
[image]
I can't confirm any problems yet.

NeoRaider, 2016-07-11:
@mokkin, which of the patches...?

mokkin, 2016-07-11:
@NeoRaider I'm using the patch from LEDE which you have included at https://github.com/freifunk-gluon/gluon/tree/ffd1f0b3a5b2950d8c626a2de26fb1dc8e384408
Version 0.8.5.1-exp is our experimental firmware including this patch. You can find it here: http://vpnb.leipzig.freifunk.net:8004/freifunk-gluon/images-freifunk-leipzig-20160511/

A-Kasper, 2016-07-24:
Ist der Patch im aktuellen 2016.1.x branch durch den Wechsel zur LEDE
Version enthalten? Wenn nicht: werden noch Tester gebraucht?!

mokkin, 2016-07-25:
@A-Kasper Ich denke nicht. NeoRaider hat ihn erstmal entfernt. Wenn du ihn wie ich testen und nutzen willst, baue einfach gluon neu mit den oben von mir verlinktem Stand. Momentan ist ja noch OpenWRT der Unterbau, soweit ich weiß. Aber ich freue mich auch schon auf LEDE.

@NeoRaider btw. it was @dangowrt, a member of our Leipzig community who had the problem with his CPE510. He told me, he was using an upper channel (116?) with high power for testing purposes. Usually we are on ch. 44. Maybe he can provide additional information.

Sprinterfreak, 2016-07-30:
This could in fact reduce the possible tx rate because other devices far away or the noise floor could possibly occupy the cpe on too much gain amplification.
We should make this as an uci option at wireless config.

Byggvir, 2016-08-12:
Guten Tag, der Patch hilft. Zwei unserer Knoten CP210 v1.1 sind über 160m durch Laubbäume miteinander verbunden. Nach dem Patch geht es von ~85dBM rauf auf ~70dBM.

Kurze Frage: Überleben die Einstellungen einen reboot?

mokkin, 2016-08-12:
Also meine CPE510 laufen äußerst stabil und haben auch schon einige reboots hinter sich.
Zur 210er kann ich nix sagen, sollte aber ähnlich sein.

viisauksena, 2016-08-12:
@mokkin von welchem patch genau redest du ? - der oben vorgeschlagene Vorgang von @Sunz3r dürfte nicht rebootfest sein - wie auch.

NeoRaider, 2016-08-12:
Der Befehle von @Sunz3r sind so definitiv inkorrekt, da sie die entsprechenden GPIOs einfach auf high setzen, anstatt sie in den speziellen LNA-Modus zu setzen - das geht nämlich nur per Kernel-Patch.

Ich hatte insgesamt 3 Patches für dieses Ticket aus LEDE gebackportet und erstmal wieder rückgängig gemacht. Ich denke, die beiden unkritischen werde ich wieder anwenden. Der letzte, der die Sendeleitung auf den CPE510 erhöht, ist wohl problematisch und kann zu Schäden am Gerät führen.

mokkin, 2016-08-12:
@viisauksena der hier: https://github.com/freifunk-gluon/gluon/tree/ffd1f0b3a5b2950d8c626a2de26fb1dc8e384408

Sunz3r, 2016-08-12:
Das hatte ich auch schon vermutet dass die nicht einfach auf High gesetzt werden dürfen ... sollte daher auch nur zum testen sein und nicht als Lösung. :) Das hätte ich wahrscheinlich explizit nochmal erwähnen müssen.
Habe bei mir (Innenstadt) die Tests auch abgebrochen da die TQ trotz besseres Signal schlechter geworden war. Aber wahrscheinlich liegt das eher an den vielen WLANs in der Umgebung.

mokkin, 2016-08-12:
Vielleicht sollte man das Ticket auch splitten, da ja anscheinend alle außer mir die 5GHz-Variante verwenden.

Sunz3r, 2016-08-12:
Ich habe drei CPE210 in Verwendung woran ich auch meine Experimente gemacht hatte.

switch42, 2016-08-12:
Mit wie viel mW sendet der CPE510 eigentlich jetzt? Ich konnte damals nicht mehr als 10mW einstellen.

Byggvir, 2016-08-12:
Ich verwende zwei CPE210 v1.1. Die Verbindung über 160m ist extrem instabil. Mal geht es stundenlang gut, dann bricht die Verbindung plötzlich zusammen. Die Router kommen nach einem Neustart relativ schnell wieder in Verbindung. In guten Zeiten erreicht die Strecke über Stunden 5 MBit/s. Auf dem Channel 5 habe ich inzwischen alle anderen Router verdrängt. Nur ein WDR841 funkt noch auf dem Client Netz und meshed über WLAN mit einem der beiden CPE210.

mokkin, 2016-08-12:
@switch42 
```
ibss0     ESSID: "02:44:ca:ff:ee:ee"
          Access Point: 02:44:CA:FF:EE:EE
          Mode: Ad-Hoc  Channel: 44 (5.220 GHz)
          Tx-Power: 20 dBm  Link Quality: 59/70
          Signal: -51 dBm  Noise: -95 dBm
          Bit Rate: 144.4 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11an
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

Sprinterfreak, 2016-08-12:
@mokkin
20dBm + unknown Offset (9dBi) ?!
Der Link muss Kilometer weit gehen... :-D

mokkin, 2016-08-12:
@Sprinterfreak 
ich hoffe doch :) hab bisher aber nur den halben kilometer testen können

switch42, 2016-08-13:
@mokkin Bei mir sind es nur 10dBm, mehr kann ich nicht einstellen. Kannst du mir einen Tipp geben? Gern auch per Mail an switch42(at)gmx.de

Sprinterfreak, 2016-08-13:
Naja, 20dBm (+9dBi Antenne) sind bei der CPE210 etwa 800mW EIRP...

mokkin, 2016-08-13:
Es ging hier aber um die 5GHz Variante.

NeoRaider, 2016-08-22:
61cde569ad5a49ceb5eb55871adbdc1964f90374 adds the unproblematic patches again, please test.

A-Kasper, 2016-08-22:
Will this also be backported to 2016.1.x?

NeoRaider, 2016-08-22:
@A-Kasper: Depending on the feedback, it's possible.

FFS-Roland, 2016-08-23:
After having read this thread from Freifunk Gera-Greiz [https://www.freifunk-gera-greiz.de/forum/-/message_boards/message/62361](url) I set on my CPE210 v1.0 Freifunk-Node only GPIO sw18 which gives very good results up to now. Setting both sw18 + sw19 will result in high sensitive but instable WLAN connections.

NeoRaider, 2016-08-23:
@FFS-Roland, as mentioned before, manually setting the GPIOs doesn't do the right thing. The GPIOs must be set to the special LNA mode, which is only possible with a kernel patch.

Sprinterfreak, 2016-08-23:
To actually better understand whats really going on. What is the difference
between activating LNA by GPIO and by kernel patch?
I suggest simply cranking up the gain by turning on the hardware LNA
without adjusting the drivers squelch profile results in an overload
because of the driver is occupied by analyzing background noise rather then
actual signals?

NeoRaider, 2016-08-23:
The LNA mode is a special mode of the GPIO which makes the WLAN adapter drive the pins adaptively. This is a feature of the Atheros SoC, I'm not sure about the details...

FFS-Roland, 2016-08-23:
Sometimes theory an practise seem to different worlds.  The experience of Mathias and Marcus in the mentioned thread and also mine show that it works, even if I cannot tell you why ...

NeoRaider, 2016-08-23:
I'm not saying just setting the GPIO to 1 doesn't have an effect. I'm saying that it doesn't have the *desired* effect.

A-Kasper, 2016-08-24:
Am 23.08.2016 um 00:40 schrieb Matthias Schiffer:
> @A-Kasper <https://github.com/A-Kasper>: Depending on the feedback, it's
> possible.

I've tried it on a cpe210 1.0 and acpe210 1.1. Both have higher TQ and
much better dbm signal now...

A-Kasper, 2016-08-30:
As I already told: 
Signal and TQ are much better with this Fix.  Here is an example Screenshot. I'm using the firmware on 23 CPE210/510 v1.0 or v1.1. No Problems spotted.
[image]
[image]

NeoRaider, 2016-08-30:
Okay, this seems to work correctly now.

closed 2016-08-30 by NeoRaider

mokkin, 2016-09-01:
I upgraded today from the old to the new patch. TQ is the same, but the signal changed about 20dBm.
[image]
[image]

mokkin, 2016-09-02:
20dBm worse, I mean...

A-Kasper, 2016-09-02:
Am 02.09.2016 um 11:16 schrieb mokkin:
> 20dBm worse, I mean...

strange,

we are using 802.11s. but could this effect the signal strength?

mokkin, 2016-09-02:
Does anybody else use CPE510? I mean especially the 5GHz version right now.

commit ffd1f0b3 (2016-06-09): ar71xx: backport fixes for better CPE210/510 support

This patchset enables the RX LNA for the CPE210/510, improving RX by about
20dB. The profiles for CPE210 and CPE510 is split into two images, so the
CPE510 can use the correct ART offset, improving the TX power by 10dB.

Fixes #796