freifunk-gluon/gluon #798: Webif page loads with error message in setup mode

pbriesch, 2016-06-09:
Hi,

after upgrading two TP-LINK Archer C5 to v2016.1.5 using the Webif in the config mode I get the following error message in my webbrowser when I want to go later to the fastd setup page:

```
/usr/lib/lua/luci/dispatcher.lua:364: Failed to execute cbi dispatcher target for entry '/admin/mesh_vpn_fastd'.
The called action terminated with an exception:
/usr/lib/lua/luci/util.lua:311: bad argument #1 to 'pairs' (table expected, got string)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:364: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:135: in function </usr/lib/lua/luci/dispatcher.lua:134>
```

Please let me know if you need further information.

Thanks, Paul

NeoRaider, 2016-06-09:
While the error message doesn't really say a lot, I presume the issue is caused by a broken `/etc/config/fastd`. Is it possible that your configuration section `mesh_vpn` contains `option method ...` instead of `list method ...`? Have you ever changed that value by other means than the web interface?

pbriesch, 2016-06-16:
No, I haven't changed that value by other means. I'm going to upgrade the router with the same version again, however, without keeping the old settings.

NeoRaider, 2016-07-27:
I don't think there is any way for the fastd config to get into this broken state in the current code of Gluon. Please reopen if you have any more information how to reproduce this.

closed 2016-07-27 by NeoRaider