freifunk-gluon/gluon #806: CPE210v1.1 PoE LAN-port is not the uplink port

Gammelknup, 2016-06-23:
at CPE210v1.1 ports are changed, so uplink ist not the POE port, but the LAN port (at the gluon master build at 11.06.2016)
for me works:

uci set network.mesh_lan.ifname='eth0.2'
uci set network.wan.ifname='eth0.1'
uci commit
reboot

NeoRaider, 2016-06-23:
These commands might further break the network config (this won't survive updates, and you assume that mesh-on-lan is enabled).

So on your CPE210, the "main" port is LAN and "secondary" is WAN?

NeoRaider, 2016-06-23:
Ah, "main" and "secondary" are the UBNT names. Were they called "lan0" and "lan1" on the labels of the CPE210?

Gammelknup, 2016-06-23:
Yes they are labeled as:
- "LAN0 (POE IN)" 
- "LAN1"
https://goo.gl/dBJJS1

NeoRaider, 2016-06-23:
Ah, the issue has been caused by ffd1f0b3a5b2950d8c626a2de26fb1dc8e384408, which has been reverted already. The cpe510 and cpe210 profiles were split in the commit, but Gluon's interface selection logic was not adjusted. Everything should be fine with the current master.

closed 2016-06-23 by NeoRaider