freifunk-gluon/gluon #807: Disable CONFIG_KERNEL_KALLSYMS?

T-X, 2016-06-28:
Since we are a little short on RAM for 32MB devices, I was wondering whether we should remove the kernel symbol table from the images by unsetting CONFIG_KERNEL_KALLSYMS. Not 100% sure, but I think this had actually been the default in OpenWRT some time ago.

At least for the uncompressed vmlinux for the ar71xx-generic target I'm getting a size reduction from 3836356 bytes down to 3606900. For kernel modules there should be a similar reduction in size.

The downside of all this are less comfortable call traces, of course.

NeoRaider, 2016-07-27:
I think proper backtraces are too valuable to disable the kernel symbols.

In any case, the uncompressed numbers aren't of much value, do you have numbers for LZMA-compressed kernels?

NeoRaider, 2017-02-23:
As written before, I don't think this is a good place to shave off some bytes.

closed 2017-02-23 by NeoRaider