freifunk-gluon/gluon #811: Missing build dependency: hexdump

Tobi042, 2016-07-01:
It seems that the build process for the x86 images (we tested with x86-kvm_guest but suspect all x86 images using grub) use an executable called hexdump that is not installed by default by the packages given in the documentation.
The dangerous thing about this is, that if the hexdump executable is missing, the builds do not fail, but rather continue silently. This seems to lead to x86 images being built with defunct Grub configurations, which in turn leads to errors during boot, such as "Waiting for root device PARTUUID=-02...".  @mar-v-in should be able to provide you with a more detailed account as to why it fails. Anyway, the result is a defective virtual machine or offloader.
The executable is available in the package bsdmainutils. I recommend you include this package in the documentation.

Adorfer, 2016-07-01:
that would explain a lot about our first futro-build "bricks"

RalfJung, 2016-07-02:
>  I recommend you include this package in the documentation.

In addition, the build should check for hexdump and *fail* loudly if the binary is missing.

commit 0676f5f3 (2016-07-03): x86: stop relying on hexdump for image build signature

Fixes #811
commit 3b916a55 (2016-07-03): x86: stop relying on hexdump for image build signature

Fixes #811
closed 2016-07-03 by NeoRaider