freifunk-gluon/gluon #813: WR841N V9 page allocation failures

Casey1979, 2016-07-03:
The Memory Pressure Problems on this https://map.hamburg.freifunk.net/#!v:m;n:e8de27b77124 TL-WR841N v9 still exist.

This Bugfix helps in that way that it prevents most reboots but doesn't resolve the Problem completely.
Update kernel code for the QCA953x - Might improve stability of the TP-Link TL-WR841N/ND v9

The Failures are sometimes showing up when the Neighbours are using the builtin Wifijammer of their 2.4GHz AV-Transmitter.
Sometimes with short, medium or long traces in the Logs or even reboots.
The last one was fitting completely inside the Logbuffer.

rotanid, 2016-08-22:
seems to be a duplicate of #753, please contribute there.

Casey1979, 2016-08-22:
Even my TL-WDR4900 got shot into a reboot once by the Jammer so i can assure you that this issue ins in no way related to Problems caused by Networksize.
This is about big wifipackets consuming all Ram because there seems to be no limit on Traffic buffered.

rotanid, 2016-08-22:
ok, but you haven't said that before ;-)
reopening.

viisauksena, 2016-10-03:
if you are right (which seems likely since i got this error/hint in dmesg a lot) - than at least i can confirm this v9 v10 v11 and cpe210v1 also. with all i got mostly on daytime (which means many client in whole network and/or bigger usage of nodes) and mostly when many links involved (at ibss0)
newest example cpe - http://www.heypasteit.com/clip/2ZCB
newest example 841 - http://www.heypasteit.com/clip/2ZCG
(note that these havent failed completly, so there may be another reason also for hard failing nodes)

Casey1979, 2016-10-15:
And its under heavy attack from the neighbours AV-links integrated jammer again.
http://pastebin.com/zRxvCe6X
The only way to prevent it from directly rebooting is sysctl -w vm.panic_on_oom=0 while providing aditional cover with iperf through an accesspoint to a client.

Normal wäre das ein Fall für die Bundesnetzagentur wegen Betrieb eines Störsenders und ein Strafverfahren wegen "§ 303b Computersabotage"
"(1) Wer eine Datenverarbeitung, die für einen anderen von wesentlicher Bedeutung ist, dadurch erheblich stört, dass er"
"2. 	Daten (§ 202a Abs. 2) in der Absicht, einem anderen Nachteil zuzufügen, eingibt oder übermittelt"

Die Folgen dieses asozialen Verhaltens technisch soweit abzumildern das keine Neustarts mehr auftreten oder Prozesse abgeschossen werden wäre mir aber wesentlich lieber als mit der schweren Artillerie das Recht gegen Lernresistente durchzusetzen.

T-X, 2016-10-15:
Can someone explain a little more how this "AV-jammer" operates? Is it heavy multicast traffic (what exactly? ports, what kind of service, do you have a name of the application?) send to a Gluon nodes AP interface @1Mbit/s multicast rate? Or something else? If you don't know, then a tcpdump would be nice and maybe some "screenshots" of "H.O.R.S.T" (https://github.com/br101/horst).

Casey1979, 2016-10-17:
I tried to capture the offending transmission with kali inux via netsniff-ng on a 3-stream usbadapter and a search for packets bigger than 1600 bytes but could not find anything near "size 1974".
A tcpdump on a router with enough flash revealed nothing either.
Yet the WR841N and the WR802N reboot with the usual messages after i heard the neighbours jumping down the stairs on their run to their av-transmitter.
The load is then that heavy that even the wifi led stops blinking until it finally reboots or the neighbours let go fast enough so that it is able to recover itself.
Bigger devices are also affected but can cope with it without any bigger impacts than a degraded linkscore and higher loadaverage at least 99.9% of the time.
Another observation, enabling adhoc and 802.11s more than doubles the chance for a reboot.
Is there a way to make ath9k print out this size 1974 packet it failed to allocate the skb for or to drop skbs with size 1974 directly without further processing?
Either way, the kernel should never allocate so much space for networktraffic that it runs into oom situations.

rotanid, 2016-10-17:
so this av-transmitter has nothing to do with regular wifi and especially isn't connected to your gluon-routers?
it uses the same frequency and seems to not comply with standards, though?
If your neighbours are ignorant when explaining the problem to them, I guess you could call Bundesnetzagentur so they drive by with their measuring car...

T-X, 2016-10-17:
@Casey1979: The kernel often allocates skb data with some extra bytes before and after the actual packet. To be prepared for potential additions of headers without needing to reallocate new and copy memory. So even if ath9k says 1976, you are probably not looking for that amount within wireshark/tcpdump/etc.

Wireshark has a nice IO Graph feature and other nice things in the Statistics tab (summary of capture is useful, too). Can you name a rough amount of UDP vs. TCP data?

T-X, 2016-10-17:
Btw., if you can't detect the "AV-jammer" with Linux tools, how were you able to know about and pin-point the issue to this device of your neighbors?

Have you seen it, do you know the name of this device or application?

Casey1979, 2016-10-17:
@rotanid that av-transmitter is not an issue as long as it is transmitting is transmitting its tv signals but wifi is a problem to that transmitter, thats why they constantly run to the transmitter and then the 32mb ram routers are starting to fill their logs and finally reboot. can you explain the reboots at exactly these moments if the transmitter is not the issue?
my guess was that it is constantly transmitting a copy of the signal it receives at keypress, but that still doesnt explain the relatively big skb size of 1974 that would explain the mass of duplicate replies i received to a ping one time tho.
the Bundesnetzagentur is an option as that transmitter must be transmitting at around 100mW while SDRs at 2.4ghz are only allowed to transmit at 10mW and they also have a transmitter at 5GHz channel 44 with at around 100mW while 5GHz SDRs are only allowed 25mW and only over 5,725 GHz.
guess what channels freifunk hamburg is using, yes 1 and 44 ...
that would solve the DVB-T interference coming from them too, but is not too much friendlier than taking them to court.
and please explain to me too why the kernel should use up all of its ram to buffer traffic (wheter it maybe rx or tx) when it is clear that this leads to oom conditions with process kills and reboots.
i would provide a dump so you could see yourself, but that is not possible either for privacy reasons.

@T-X after experiencing problems with the first freifunk router (wr841nd V9) 18 monts ago i started my investigation by running an iperf test over a lancom l-54ag on channel 1 which seems to be immune to the transmitter and observed where the action started, it was that notoriously stupid and egomaniacal neighbour at his sat-antenna, because he is to lazy to lay antennacables to his kids in the attic and into the living room, then some loud talk that goes through the walls and the constant running down the stairs to the kitchen where that cable from outside ends, my guess is that they must be at around 5k to 8k runs to the transmitter already.
And no, i can't investigate there further because that neighbour just doesn't see the problems he is causing.
at one time i filled both channel 1 and 44 for a full week with iperf traffic (nothing illegal done), guess what, after two days running around along with several furious outbreaks they sat 1 week in the kitchen looking tv, no progress, nothing learned. it would be funny if it wasn't that sad ...
i don't want to quit freifunk but wouldn't want to harm that already broken family further, that is why i am searching for a technical solution to solve this issue from my side without causing legal trouble.

T-X, 2016-10-17:
@Casey1979: I'm having trouble to follow. So you are saying these AV-transmitters are not wifi but SDR based with their own protocol but on the same frequency?

If that were the case, then their transmissions would never end up as skbs in the ath9k driver or anywhere in the Linux kernel as the wifi chips would not be able to decode them...

Casey1979, 2016-10-17:
@T-X they don't cause any trouble as long as it doesn't try to jam the channel which happens only after one of the neighbours arrived at the transmitter which can be clearly heard because they are jumping down the stairs.
no one at their home = no problems other than higher ANI levels between 3 and 6.

T-X, 2016-10-17:
@Casey1979: You are saying that the only hint and correlation you have and can share is that you hear noises briefly before your router crashes?

Casey1979, 2016-10-17:
@T-X No that is not what i explained to you but just one of the many many points someone gets after 18 months of observation and personal experience which also included a hanging speedport wifi printer when they were on channel 1 which i just didn't knew but experienced for around 9 years already.

Should i close this ticket as i can't expect anything helpful anyway although viisauksena seems to be experiencing the same trouble?

closed 2016-10-17 by T-X

T-X, 2016-10-17:
PS: The only thing I could think of is an occupied channel causing the router to queue too many of its own packets for transmission. If you can reproduce such a case with some jamming device on your own (maybe a Freifunk router close to a microwave? Another, but separate AP on channel 1 with iperf'ing many UDP multicast packets?), then feel free to reopen. Or better, as @rotanid suggested, append to #753, as the latest posts there indicate some potential queueing issues, too.

Thanks for your investigations and reporting so far, though, @Casey1979.