freifunk-gluon/gluon #815: WR940N v3 eth0 freezing

H4ndl3, 2016-07-04:
I encountered a strange bug. At first gluon is working fine, the vpn connection to the gateway is stable and I can connect via ssh to the WR940N using its IPv4 assigned by my main router. However, after an unspecified amount of time (or maybe it's triggered by a certain event?) eth0 seems to freeze. The WR940N becomes unreachable from the outside. I get ping timeouts I'm using its local IPv4 or its IPv6. Wifi is still working. When I then connect via ssh and run `ifconfig eth0 down` followed by `ifconfig eth0 up` everything goes back to normal until eth0 seems to freeze again.

What I've done so far:
- changed the ethernet cable
- got a new WR940N v3 in case the old one was broken somehow
- changed the ethernet port of my main router
- used a WR840ND v8 (which worked smoothly) to rule out some kind of incompatibility between my main router and gluon devices

A `logread` gives me this output:

> Sun Jul  3 22:59:31 2016 kern.info kernel: [  197.050000] ibss0: Creating new IBSS network, BSSID 02:d1:11:37:fc:84
Sun Jul  3 22:59:31 2016 kern.info kernel: [  197.050000] IPv6: ADDRCONF(NETDEV_CHANGE): ibss0: link becomes ready
Sun Jul  3 22:59:33 2016 kern.info kernel: [  198.890000] br-client: port 2(client0) entered forwarding state
Sun Jul  3 22:59:41 2016 kern.info kernel: [  207.110000] device client0 left promiscuous mode
Sun Jul  3 22:59:41 2016 kern.info kernel: [  207.120000] br-client: port 2(client0) entered disabled state
Sun Jul  3 22:59:42 2016 daemon.notice netifd: radio0 (2253): Configuration file: /var/run/hostapd-phy0.conf
Sun Jul  3 22:59:42 2016 kern.info kernel: [  207.790000] IPv6: ADDRCONF(NETDEV_UP): client0: link is not ready
Sun Jul  3 22:59:42 2016 kern.info kernel: [  207.810000] device client0 entered promiscuous mode
Sun Jul  3 22:59:42 2016 kern.info kernel: [  207.810000] br-client: port 2(client0) entered forwarding state
Sun Jul  3 22:59:42 2016 kern.info kernel: [  207.820000] br-client: port 2(client0) entered forwarding state
Sun Jul  3 22:59:42 2016 daemon.notice netifd: radio0 (2253): client0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Jul  3 22:59:42 2016 daemon.notice netifd: radio0 (2253): Using interface client0 with hwaddr a6:2d:b1:f0:2a:ce and ssid "Freifunk"
Sun Jul  3 22:59:42 2016 daemon.notice netifd: radio0 (2253): client0: interface state COUNTRY_UPDATE->ENABLED
Sun Jul  3 22:59:42 2016 daemon.notice netifd: radio0 (2253): client0: AP-ENABLED
Sun Jul  3 22:59:42 2016 kern.info kernel: [  207.850000] IPv6: ADDRCONF(NETDEV_CHANGE): client0: link becomes ready
Sun Jul  3 22:59:42 2016 kern.info kernel: [  208.090000] IPv6: ADDRCONF(NETDEV_UP): ibss0: link is not ready
Sun Jul  3 22:59:42 2016 kern.info kernel: [  208.110000] ibss0: Created IBSS using preconfigured BSSID 02:d1:11:37:fc:84
Sun Jul  3 22:59:42 2016 kern.info kernel: [  208.120000] ibss0: Creating new IBSS network, BSSID 02:d1:11:37:fc:84
Sun Jul  3 22:59:43 2016 kern.info kernel: [  208.130000] IPv6: ADDRCONF(NETDEV_CHANGE): ibss0: link becomes ready
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Network device 'client0' link is up
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Network device 'ibss0' link is up
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Interface 'ibss_radio0' is enabled
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Interface 'ibss_radio0' has link connectivity
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Interface 'ibss_radio0' is setting up now
Sun Jul  3 22:59:43 2016 kern.info kernel: [  208.270000] batman_adv: bat0: Adding interface: ibss0
Sun Jul  3 22:59:43 2016 kern.info kernel: [  208.270000] batman_adv: bat0: Interface activated: ibss0
Sun Jul  3 22:59:43 2016 daemon.notice netifd: Interface 'ibss_radio0' is now up
Sun Jul  3 22:59:44 2016 kern.info kernel: [  209.820000] br-client: port 2(client0) entered forwarding state
Sun Jul  3 23:43:30 2016 daemon.notice netifd: client (1651): cat: write error: Broken pipe
Mon Jul  4 02:36:41 2016 daemon.notice netifd: client (1651): cat: write error: Broken pipe
Mon Jul  4 07:54:55 2016 daemon.notice netifd: client (1651): cat: write error: Broken pipe
Mon Jul  4 08:11:54 2016 daemon.info hostapd: client0: STA ac:5f:3e:25:a7:db IEEE 802.11: authenticated
Mon Jul  4 08:11:54 2016 daemon.info hostapd: client0: STA ac:5f:3e:25:a7:db IEEE 802.11: associated (aid 1)
Mon Jul  4 08:13:05 2016 daemon.warn td-client: Tunnel has timed out, closing down interface.
Mon Jul  4 08:13:05 2016 daemon.notice netifd: Network device 'mesh-vpn' link is down
Mon Jul  4 08:13:05 2016 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity loss
Mon Jul  4 08:13:05 2016 kern.info kernel: [33410.330000] batman_adv: bat0: Interface deactivated: mesh-vpn
Mon Jul  4 08:13:05 2016 kern.info kernel: [33410.340000] batman_adv: bat0: Removing interface: mesh-vpn
Mon Jul  4 08:13:05 2016 daemon.err td-client: Connection to ausrufer.servers.freifunk-muensterland.de lost.
Mon Jul  4 08:13:05 2016 daemon.info td-client: Performing broker selection...
Mon Jul  4 08:13:05 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Mon Jul  4 08:13:05 2016 daemon.notice netifd: Interface 'mesh_vpn' is disabled
Mon Jul  4 08:13:05 2016 daemon.notice netifd: mesh_vpn (7502): ./batadv.sh: eval: line 1: can't create /sys/class/net/mesh-vpn/batman_adv/mesh_iface: nonexistent directory
Mon Jul  4 08:13:11 2016 daemon.err td-client: Hostname resolution timed out.
Mon Jul  4 08:13:11 2016 daemon.err td-client: Hostname resolution timed out.
Mon Jul  4 08:13:11 2016 daemon.err td-client: Hostname resolution timed out.
Mon Jul  4 08:13:21 2016 daemon.info td-client: Reinitializing tunnel context.
Mon Jul  4 08:13:21 2016 daemon.info td-client: Reinitializing tunnel context.
Mon Jul  4 08:13:21 2016 daemon.info td-client: Reinitializing tunnel context.
Mon Jul  4 08:13:21 2016 daemon.info td-client: Reinitializing tunnel context.
Mon Jul  4 08:13:21 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Mon Jul  4 08:13:26 2016 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds

I truncated it because at the end it's just the last lines over and over.

What's interesting to me is that the connection froze shortly after I joined the wifi with my mobile phone. However, I'm now connected with my both my laptop and my mobile phone after restarting `eth0` and everything is working smoothly so far. But from my past experience with the other WR940N I know that the connection will freeze again.

Please let me know if there's anything I can do to debug this error further so you can fix it.

NeoRaider, 2016-07-04:
Is setting the device down and up again the only way you've found to fix the issue when you encountered it? Does dis- and reconnecting the ethernet cable affect it when ethernet is broken?

How often does this occur?

H4ndl3, 2016-07-05:
Rebooting does (obviously) also fix the issue. Dis- and reconnecting the ethernet cable does also work.

Hard to tell. I have a hunch that it might be actions of wifi devices that lead to this strange behavior because just as I joined the wifi with my Samsung Galaxy S7 the connection froze again. I then dis- and reconnected the ethernet cable and now everything is back to normal again. I joined the wifi again with my S7 but now there was no freeze and no connection problems.

Logread output:

> Tue Jul  5 12:28:00 2016 daemon.warn td-client: Tunnel has timed out, closing down interface.
Tue Jul  5 12:28:00 2016 daemon.notice netifd: Network device 'mesh-vpn' link is down
Tue Jul  5 12:28:00 2016 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity loss
Tue Jul  5 12:28:00 2016 kern.info kernel: [10304.140000] batman_adv: bat0: Interface deactivated: mesh-vpn
Tue Jul  5 12:28:00 2016 kern.info kernel: [10304.170000] batman_adv: bat0: Removing interface: mesh-vpn
Tue Jul  5 12:28:00 2016 daemon.err td-client: Connection to 176.9.88.123 lost.
Tue Jul  5 12:28:00 2016 daemon.info td-client: Performing broker selection...
Tue Jul  5 12:28:00 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:28:01 2016 daemon.notice netifd: Interface 'mesh_vpn' is disabled
Tue Jul  5 12:28:01 2016 daemon.notice netifd: mesh_vpn (8911): ./batadv.sh: eval: line 1: can't create /sys/class/net/mesh-vpn/batman_adv/mesh_iface: nonexistent directory
Tue Jul  5 12:28:01 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:28:17 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:28:17 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:28:17 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:28:17 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:28:22 2016 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds
Tue Jul  5 12:28:27 2016 daemon.info td-client: Performing broker selection...
Tue Jul  5 12:28:27 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:28:27 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:28:43 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:28:43 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:28:43 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:28:43 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:28:48 2016 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds
Tue Jul  5 12:28:50 2016 daemon.warn dnsmasq[1685]: no servers found in /tmp/resolv.conf.auto, will retry
Tue Jul  5 12:28:53 2016 daemon.info td-client: Performing broker selection...
Tue Jul  5 12:28:53 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:28:53 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:29:09 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:29:09 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:29:09 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:29:09 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:29:14 2016 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds
Tue Jul  5 12:29:19 2016 daemon.info td-client: Performing broker selection...
Tue Jul  5 12:29:19 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:29:19 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:29:35 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:29:35 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 12:29:35 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 12:29:35 2016 daemon.err td-client: Failed to resolve hostname 'domaene46.servers.freifunk-muensterland.de'.
Tue Jul  5 12:29:40 2016 daemon.info td-client: Selected 176.9.88.123:20046 as the best broker.
Tue Jul  5 12:29:41 2016 daemon.info td-client: Tunnel successfully established.
Tue Jul  5 12:29:41 2016 daemon.notice netifd: Interface 'mesh_vpn' is enabled
Tue Jul  5 12:29:41 2016 daemon.notice netifd: Network device 'mesh-vpn' link is up
Tue Jul  5 12:29:41 2016 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity
Tue Jul  5 12:29:41 2016 daemon.notice netifd: Interface 'mesh_vpn' is setting up now
Tue Jul  5 12:29:41 2016 kern.info kernel: [10404.790000] batman_adv: bat0: Adding interface: mesh-vpn
Tue Jul  5 12:29:41 2016 kern.info kernel: [10404.790000] batman_adv: bat0: The MTU of interface mesh-vpn is too small (1364) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
Tue Jul  5 12:29:41 2016 kern.info kernel: [10404.820000] batman_adv: bat0: Interface activated: mesh-vpn
[... cut ...]
Tue Jul  5 15:30:30 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 15:30:30 2016 daemon.info td-client: Reinitializing tunnel context.
Tue Jul  5 15:30:30 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.
Tue Jul  5 15:30:35 2016 daemon.err td-client: No suitable brokers found. Retrying in 5 seconds
Tue Jul  5 15:30:40 2016 daemon.info td-client: Performing broker selection...
Tue Jul  5 15:30:40 2016 daemon.err td-client: Failed to resolve hostname '163.172.211.238'.

NeoRaider, 2016-07-05:
Okay, please also check the following things in the broken state:

* Does eth0 still have an IP address?
* What's the output of `ifstatus wan`?
* It would be great to know if sending or receiving packages is failing. If tcpdump-mini is too big to fit in the free space on the device, you can use `batctl td`, which provides a small and very limited tcpdump. Connecting a hub to the connection or using another Linux (OpenWrt?) system as upstream router to see the other side would be even better.

H4ndl3, 2016-07-05:
After waiting and now joining the wireless lan again with my Galaxy S7 the connection broke again, so I am somewhat able to reproduce the problem.

- eth0 doesn't have an IP address, but it also doesn't have one when everything is working fine
>eth0      Link encap:Ethernet  HWaddr A4:2B:B0:XX:XX:XX
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:403505 errors:0 dropped:0 overruns:0 frame:0
          TX packets:162851 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:69857834 (66.6 MiB)  TX bytes:52365487 (49.9 MiB)
          Interrupt:4

- `ifstatus wan`
>{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 3964,
        "l3_device": "br-wan",
        "proto": "dhcp",
        "device": "br-wan",
        "updated": [
                "addresses",
                "routes",
                "data"
        ],
        "metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "192.168.1.66",
                        "mask": 24
                }
        ],
        "ipv6-address": [
        ],
        "ipv6-prefix": [
        ],
        "ipv6-prefix-assignment": [
        ],
        "route": [
                {
                        "target": "192.168.1.1",
                        "mask": 32,
                        "nexthop": "0.0.0.0",
                        "source": "192.168.1.66\/32"
                },
                {
                        "target": "0.0.0.0",
                        "mask": 0,
                        "nexthop": "192.168.1.1",
                        "source": "192.168.1.66\/32"
                }
        ],
        "dns-server": [
        ],
        "dns-search": [
        ],
        "inactive": {
                "ipv4-address": [
                ],
                "ipv6-address": [
                ],
                "route": [
                ],
                "dns-server": [
                        "192.168.1.1"
                ],
                "dns-search": [
                ]
        },
        "data": {
                "leasetime": 86400
        }
}
- tcpdump-mini is too big for the device. How should I run `batctl td`?

NeoRaider, 2016-07-05:
Ah, of course, the IP address is on br-wan (and ifstatus shows that it is still there, and also the DNS server and everything is available).

Can you still ping 192.168.1.1 when is is broken?

You can just run `batctl td eth0`. Try to see if there are any packet originating from your upstream router.

H4ndl3, 2016-07-05:
I can't ping 192.168.1.1 when it's broken.

Will try asap.

H4ndl3, 2016-07-06:
I had to wait for a while but then those packets appeared on eth0:

>19:00:01.173315 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:02.164611 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:02.724917 IP6 fe80::a42c:b0ff:fef0:2ace.546 > ff02::1:2.547: UDP, length 113
19:00:03.164353 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:05.179521 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:06.174946 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:07.174370 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:08.183227 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:09.174362 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:10.174340 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:11.176672 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:12.174355 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:13.174343 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:14.190158 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:15.184371 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:16.184385 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:17.195517 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:18.194345 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:19.194347 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:20.202555 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:21.194371 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:22.194462 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:27.206734 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:28.204357 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:29.204358 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:32.211483 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:33.204385 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:00:34.204499 ARP, Request who-has 192.168.1.1 tell 192.168.1.66 (a6:2c:b0:xx:xx:xx), length 28
19:01:55.505045 IP6 fe80::a42c:b0ff:fef0:2ace.546 > ff02::1:2.547: UDP, length 113
19:03:57.994850 IP6 fe80::a42c:b0ff:fef0:2ace.546 > ff02::1:2.547: UDP, length 113

Anything else I can do?

NeoRaider, 2016-07-06:
Hmm, all of these packets originate from your Gluon node. It will be hard to tell without a trace from the other side if the sending or receiving side is broken.

You could try connecting a third device to the network and send broadcast frames from there, so you can see if these still reach the node. Broadcast frames don't need ARP, so you should see them even when your node can't send anything anymore.

H4ndl3, 2016-07-06:
I can connect to my main router which runs busybox via ssh or use my laptop running Ubuntu for further debugging. How can I send a broadcast frame?

NeoRaider, 2016-07-06:
If you have IPv6 (link-local is enough), you can send an all-node multicast ping using `ping6 ff02::1%ifname`, where ifname is the name of the outgoing interface of the sending system. `ping -b 192.168.1.255` is an IPv4 local broadcast ping and should also work.

H4ndl3, 2016-07-06:
I sent a magic packet from my mobile phone (which should be broadcast too afaik), a ping6 from a virtual machine behind my main router and a ping to 192.168.1.255 on my main router. I don't know if I did it right because ping on busybox has no `-b` switch and there was no answer from any device. Using ping6 I got only an answer from the virtual machine itself.

However, while I tried these things there were no additional packets incoming or outgoing on the WR940N other than the ones I posted above.

H4ndl3, 2016-08-16:
The problem vanished. Maybe my main router or other network components where at fault, I don't know.

closed 2016-08-16 by H4ndl3