freifunk-gluon/gluon #818: Disable fastd if mesh on wan gets enabled

rubo77, 2016-07-08:
I see no scenario where fastd should stay enebled

jplitza, 2016-07-08:
Why wouldn't you have a small subnet where multiple nodes mesh via cable and simultaneously a router offers one of the nodes internet access?

NeoRaider, 2016-07-08:
The normal usecase is having multiple nodes with mesh-on-WAN in a home network, and having one of them use the mesh VPN.

closed 2016-07-08 by NeoRaider

rubo77, 2016-07-08:
I thought of automatically disabling the fastd-option if it is enabled, because what I just had was this problem:

my wr4900 was connected to the interne with fastd enabled and Mash-on-LAN

my 1043 had Mesh-on-WAN enabled and was connected via kable to the 4900.

Since the 1043 had also fastd enabled it created another fastd tunnel through the 4900 - a tunnel inside the tunnel.

I see no useful usecase where this should be possible?

jplitza, 2016-07-09:
Yes, VPN tunnels should never be established through Freifunk. But a more appropriate countermeasure would be to block forwarding of VPN traffic on the nodes or on the servers.

rubo77, 2016-07-09:
that would be a new issue? "block forwarding of VPN traffic on the nodes"

NeoRaider, 2016-07-09:
Doing this on the VPN servers is much more appropriate (and simpler).

Also, mesh-on-lan and mesh-on-wan are expert mode features. Users are expected to know what they are doing when using these features; "magically" changing other, mostly unrelated settings is not acceptable for the expert mode.