freifunk-gluon/gluon #820: error message in log when disabling mesh-vpn

rotanid, 2016-07-08:
```
Fri Jul  8 17:42:24 2016 daemon.notice fastd[1375]: terminating fastd
Fri Jul  8 17:42:24 2016 daemon.info fastd[1375]: deleting peer <mesh_vpn_backbone_peer_gw01>
Fri Jul  8 17:42:24 2016 kern.info kernel: [13154.330000] batman_adv: bat0: Interface deactivated: mesh-vpn
Fri Jul  8 17:42:24 2016 kern.info kernel: [13154.370000] batman_adv: bat0: Removing interface: mesh-vpn
Fri Jul  8 17:42:24 2016 daemon.notice netifd: Network device 'mesh-vpn' link is down
Fri Jul  8 17:42:24 2016 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity loss
Fri Jul  8 17:42:24 2016 daemon.notice netifd: Interface 'mesh_vpn' is disabled
Fri Jul  8 17:42:24 2016 daemon.notice netifd: mesh_vpn (24816): ./batadv.sh: eval: line 1: can't create /sys/class/net/mesh-vpn/batman_adv/mesh_iface: nonexistent directory
```

jplitza, 2016-07-09:
You are probably referring to the last line? That doesn't indicate any error and in fact was anticipated by the developers of the batadv netifd module, as the `|| true` suggests:
```sh
proto_batadv_teardown() {
	local config="$1"
	local iface="$2"

	echo "none" > "/sys/class/net/$iface/batman_adv/mesh_iface" || true
}
```
So if this only manifests itself as an error message in the logs, simply ignore it. (Maybe we should start an FAQ with "These log messages are normal, please don't report them")

NeoRaider, 2016-07-09:
This ticket was created in reaction to an IRC conversation. Even though the message can be ignored, making it disappear would be prefereable (but it's obviously an upstream issue in the batman-adv package).

rotanid, 2016-09-08:
according to @NeoRaider on IRC the script batadv.sh isn't used anymore and therefore this issue can be closed for gluon.

closed 2016-09-08 by rotanid

NeoRaider, 2016-09-08:
Fixed as a side-effect of d1e6dfb7e3a02ede79224fbce1a3e9772ed81471