freifunk-gluon/gluon #821: rare fail of autoupdater "brick" tp 841 (only tftp left)

viisauksena, 2016-07-09:
on some of our tp841v9 and v10 we observe that they sometimes fail to autoupdate and beeing stuck in boot. Only powerLED is on, nothing more happens.
While the autoupdater normaly goes fine we dont know yet the reasons - 
maybe it was because some RAM issues, maybe Load, maybe lousy connection ...

seen are these on updates v2016.1.xy up to the latest one - no pattern at all

sorry that we are not able to give more info right now, we will have to collect these routers and do a serial  console check ... 
tftp worked fine to bring some of them back ...

2tata, 2016-07-17:
Could you tell us how many Routers do you have in your Freifunk Network? On Freifunk Nordwest we had over 1000 of Mesh Router in one layer 2 network so the routing tables had high RAM consumption. The result was, while flashing the new image getting over the autoupdater bricks some routers with RAM carapaces of 32MB. At the moment I work on a patch to kill the batman-adv network interface before flashing the new image. This will have the effect of removing all batman-adv routing tables of the RAM.

viisauksena, 2016-07-18:
i suspect that this may be the case, low ram which isnt enough to finish sysupgrade ... we have 8 supernodes/gw and every of our 400+- routers connect to 2 of them (so there will be 2 entrys for euch originator how this could be reached) ... 
i see a rate of some low one digit percentage of routers who did not make it ... and it seems that the router is crashing while sysupgrade was running. 
this is heavy guessing , we thougt of making a reboot before an upgrade but not sure if our observation and/or this solution are enough. 
there may be also lousy power adapters a reason (the ones which make noise ...)
the main problem is that almost never we have this problem with a router right in front of us. Mostly many many many sysupgrade at home didnt act like this. While we wasnt aware of this behaviour we maybe thought that something else happend in this cases (like unplug to early or whatever).

Casey1979, 2016-07-18:
I can confirm reboots for 32MB devices because of ram consumption under heavy incoming wifitraffic after the firmwareimage is downloaded to the Node.
But i didn't try automatic updates only manual flashing.
Usually the Node reboots shortly after finishing its download and right before the flashcommand or it doesn't and is successfully flashing its firmwareimage.
So my suggestion is to wait a few seconds between downloadcompletion and the flashcommand so that it is able to fail and reboot without firmwarecorruption.
It may still fail while flashing but the chance is much lower if it survived a moment with the downloaded image in ram.
For maximum chance of successfull flash download the image shut down all networkinterfaces wait and then flash.

2tata, 2016-07-18:
Hi,

> Usually the Node reboots shortly after finishing its download and right before the flashcommand or it doesn't and is successfully flashing its firmwareimage.
> So my suggestion is to wait a few [...]

Exactly on this point after Finished the Download process, my patch will drop the batman-adv Routing tables.
This makes around 3MB of RAM free on a mesh network with ~150 Routers.
So on bigger mesh networks it will clear the RAM significantly.

cheers
Tarek

rotanid, 2016-07-18:
@2tata are you planning on providing this patch to the public or even doing a pull request to gluon for it?

2tata, 2016-07-18:
On 07/19/16 00:27, Andreas Ziegler wrote:
> @2tata are you planning on providing this patch to the public or even doing a pull request to gluon for it?

You can find this patch now on the Mailinglist

cheers
Tarek

viisauksena, 2016-07-19:
this is worth mentioning https://github.com/freifunk-gluon/gluon/commit/8f5a7c90019085dd8f02f96ab0e7a905f82e1be0

rotanid, 2016-08-22:
this could be fixed with the recent updater improvements in the master branch, feel free to reopen it if you still have this problem while running the latest master commit

closed 2016-08-22 by rotanid