freifunk-gluon/gluon #830: Probleme beim Bauen von Unify AP AC Pro Images

RubenKelevra, 2016-07-25:
Ich wollte gern aus dem Master Images für die Unify AP AC Pros bauen, leider wird nur ein Image für "Sysupgrade" generiert. Im Ordner Factory liegen keine Images für diesen Typ Hardware.

Ich hab via TFTP probiert auf einen das Sysupgrade-Image zu spielen, leider frisst er diese nicht.

Ist das ein noch ein bekanntes Kompabilitätsproblem oder liegt hier noch ein Fehler im Buildscript vor? :)

NeoRaider, 2016-07-25:
Die Unifi AP AC Pro haben im "regulären" Updater eine Signaturprüfung, daher können wir keine sinnvollen Factory-Images erzeugen.

Die Stock-Firmware enthält allerdings auch das Tool mtd, mit diesem kann das sysupgrade-Image direkt in den Flash geschrieben werden (siehe https://wiki.openwrt.org/toh/ubiquiti/unifiac ).

RubenKelevra, 2016-07-25:
Bei den bisherigen Unifys konnte man die mit dem erzeugten Image via TFTP flashen, prüft TFTP diese Signatur auch? Oder reicht da vielleicht doch ein Magic-Byte? :)

NeoRaider, 2016-07-27:
Wenn ich mich richtig erinnere, habe ich auch im Dump vom Uboot Strings gefunden, die auf eine RSA-Signaturprüfung hindeuten. Ist allerdings eine Weile her, dass ich mir das angeschaut habe, ich könnte mich irren.

Das neue Factory-Format ist etwas anders als das alte (darüber hinaus, dass es jetzt eine Signatur am Ende gibt), und da die Sysupgrade-Images benutzbar sind, werde ich keine Zeit in die Analyse des neuen Formats stecken.

RubenKelevra, 2016-07-29:
Ich seh grade, das OpenWRT-Wiki gibt eine Anleitung für TFTP, können wir nicht einfach das OpenWRT-Layout benutzen, was offenbar ja funktioniert? :) 

Oder übersehe ich etwas?

[image]

NeoRaider, 2016-07-29:
@RubenKelevra, die Anleitung ist für die UAP Pro, nicht den UAP AC Pro.

RubenKelevra, 2016-07-29:
Sorry. Mehr Kaffee. ;)

RubenKelevra, 2016-08-03:
Denke das Problem ist gelöst, zwei Geräte laufen super stabil bei uns mit der 2016.1.x mit backportiertem Patch :)

Bei Interesse:

https://github.com/VfN-NRW/gluon-firmware-builder/blob/master/gluon08-repo-patches/device-support/0001-add-support-for-unifi-ac-pro-lite.patch

Danke & Gruß

Ruben

closed 2016-08-03 by RubenKelevra