freifunk-gluon/gluon #833: NTP client does not rotate through rrset

mweinelt, 2016-07-28:
We accidentally had an IPv4 address in our `ntp.services.ffda.io` hostname in addition to the IPv6 address. And it seems like the ntp client does not rotate through the rrset, since always the same address was retried.

```
# nslookup ntp.services.ffda.io
Server:    127.0.0.1
Address 1: 127.0.0.1 localhost

Name:      ntp.services.ffda.io
Address 1: fd01:67c:2ed8:a::53:1
Address 2: 10.223.254.53
```

```
root@64289-rocket-cloud:~# ps | grep ntp
 1639 root      1524 S    /usr/sbin/ntpd -n -S /usr/sbin/ntpd-hotplug -p ntp.services.ffda.io
 2426 root      1520 S    grep ntp
root@64289-rocket-cloud:~# strace -p1639
Process 1639 attached
restart_syscall(<... resuming interrupted call ...>) = 0
gettimeofday({1469669437, 327510}, NULL) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(3, SOL_IP, IP_TOS, [16], 4)  = 0
gettimeofday({1469669437, 328763}, NULL) = 0
sendto(3, "#\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, MSG_DONTWAIT, {sa_family=AF_INET, sin_port=htons(123), sin_addr=inet_addr("10.223.254.53")}, 16) = -1 ENETUNREACH (Network is unreachable)
write(2, "ntpd: send failed: Network is un"..., 42) = 42
close(3)                                = 0
poll(0x9a61a8, 0, 33000^CProcess 1639 detached
 <detached ...>
root@64289-rocket-cloud:~# strace -p1639
Process 1639 attached
restart_syscall(<... resuming interrupted call ...>) = 0
gettimeofday({1469669503, 357426}, NULL) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(3, SOL_IP, IP_TOS, [16], 4)  = 0
gettimeofday({1469669503, 358649}, NULL) = 0
sendto(3, "#\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, MSG_DONTWAIT, {sa_family=AF_INET, sin_port=htons(123), sin_addr=inet_addr("10.223.254.53")}, 16) = -1 ENETUNREACH (Network is unreachable)
write(2, "ntpd: send failed: Network is un"..., 42) = 42
close(3)                                = 0
poll(0x9a61a8, 0, 33000)                = 0 (Timeout)
gettimeofday({1469669536, 387469}, NULL) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(3, SOL_IP, IP_TOS, [16], 4)  = 0
gettimeofday({1469669536, 388627}, NULL) = 0
sendto(3, "#\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, MSG_DONTWAIT, {sa_family=AF_INET, sin_port=htons(123), sin_addr=inet_addr("10.223.254.53")}, 16) = -1 ENETUNREACH (Network is unreachable)
write(2, "ntpd: send failed: Network is un"..., 42) = 42
close(3)                                = 0
poll(0x9a61a8, 0, 33000)                = 0 (Timeout)
gettimeofday({1469669569, 416820}, NULL) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(3, SOL_IP, IP_TOS, [16], 4)  = 0
gettimeofday({1469669569, 417992}, NULL) = 0
sendto(3, "#\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, MSG_DONTWAIT, {sa_family=AF_INET, sin_port=htons(123), sin_addr=inet_addr("10.223.254.53")}, 16) = -1 ENETUNREACH (Network is unreachable)
write(2, "ntpd: send failed: Network is un"..., 42) = 42
close(3)                                = 0
poll(0x9a61a8, 0, 33000)                = 0 (Timeout)
gettimeofday({1469669602, 446833}, NULL) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
bind(3, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(3, SOL_IP, IP_TOS, [16], 4)  = 0
gettimeofday({1469669602, 448168}, NULL) = 0
sendto(3, "#\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, MSG_DONTWAIT, {sa_family=AF_INET, sin_port=htons(123), sin_addr=inet_addr("10.223.254.53")}, 16) = -1 ENETUNREACH (Network is unreachable)
write(2, "ntpd: send failed: Network is un"..., 42) = 42
close(3)                                = 0
poll(0x9a61a8, 0, 33000d^CProcess 1639 detached
 <detached ...>
```

NeoRaider, 2016-07-28:
Busybox limitations, great...

christf, 2016-11-22:
is this the right place for this report if this is a limitation upstream?

mweinelt, 2016-11-23:
It's the right place to keep track of this limitation, but indeed we should file an upstream bug.

rotanid, 2017-11-15:
someone else reported this in the busybox bugzilla tracker on 2017-10-27.
https://bugs.busybox.net/show_bug.cgi?id=10466
it has already been fixed, but there's no release containing the fixes yet.
the fixes consist of three patches/commits:
https://git.busybox.net/busybox/commit/networking/ntpd.c?id=a1e9bc6876f6539805095e9395c604d3675c526d
https://git.busybox.net/busybox/commit/networking/ntpd.c?id=6a4f2231221c2c4f7ca82f081f442e31a6cfd051
https://git.busybox.net/busybox/commit/networking/ntpd.c?id=d5c1482fbac71c51e3add52632cdf1f9f9e6661b
@NeoRaider , how do we get these fixes into LEDE/Gluon?

rotanid, 2018-01-19:
buysbox release 1.28.0 (05.01.2018) contains the fixes.
there's no integration in OpenWrt, yet.

rotanid, 2018-02-13:
there's a pull request now to include busybox 1.28.0 in OpenWrt so this will be fixed upstream, soon.
and therefore in gluon-next, too.
https://github.com/openwrt/openwrt/pull/733

rotanid, 2018-06-05:
this has been merged in Gluon master in April and will therefore be part of the openwrt 18.06 release.
as this is not a big issue, we won't backport this.
closing as fixed.

closed 2018-06-05 by rotanid