freifunk-gluon/gluon #835: [ipq806x] TP-Link C2600 Unterstützung

RubenKelevra, 2016-07-28:
Bin ziehmlich weit gekommen, aber leider fällt es beim Image zusammenkopieren auf die Schnauze. Hab meine Patches an den @NeoRaider geschickt, der kümmert sich in den nächsten 1-2 Wochen drum. :)

Ticket hier ist nur, damit nicht jemand parallel Arbeit darein investiert.

bitnukl, 2016-08-15:
Gibt es schon etwas neues?
Ich hätte hier auch einen C2600, den es nach freier Firmware dürstet. :-)

RubenKelevra, 2016-08-15:
Von meiner Seite leider nicht, da muss echt der @NeoRaider mal nachschauen woran es Hakt, ich find den Bug leider nicht.

@kaechele hast du vielleicht Zeit mal reinzuschauen?

kev1904, 2016-08-15:
@NeoRaider wie lange dauert das noch etwa ?

NeoRaider, 2016-08-15:
Ich bin diese Woche noch mit GSoC beschäftigt, und wahrscheinlich werde ich danach erstmal an anderen Ecken weiterarbeiten, damit wir möglichst bald Gluon 2016.2 releasen können.

Nach 2016.2 werden wir Gluon auf LEDE umstellen, sodass wir ipq806x dann "umsonst" bekommen.

RubenKelevra, 2016-08-15:
@NeoRaider OpenWRT hat ja auch ipq806x-Support, das ist auch nicht das Problem, sondern der Support von dem C2600 in Gluon, selbst die Plattform ipq806x wird nach den Patches komplett richtig gebaut. Es handelt sich vermutlich nur um eine irgendwo nicht richtig gefüllte Variable in den Gluon-Makefiles.

Wir haben derzeit ein halbes Duzend Geräte ungenutzt rumstehen, weil wir auf schnellen Support gehofft haben. :|

NeoRaider, 2016-08-15:
Gluon basiert aktuell auf OpenWrt CC, das noch kein ipq806x unterstützt. Und jetzt viel Zeit für den Backport eines ganzen Targets aufzuopfern, wenn es in ein paar Wochen auch einfach so läuft, halte ich nicht für allzu zielführend...

RubenKelevra, 2016-08-16:
@neoraider

Naja das Backporting hab ich ja schon erledigt, die Patches die ich dir geschickt hab bauen erfolgreich unter OpenWRT CC ein Image für den C2600.

Nur der Support von Gluon für die ipq806x/den C2600 macht mir noch Kopfschmerzen.

Daher wäre deine Arbeit keine Vertane Zeit, da der Support von Gluon für den ipq806x/den C2600 sowieso geleistet werden muss, wenn wir auf LEDE gewechselt haben.

Davon ab geht es ja nicht um ein paar Wochen, da wir schlecht den Wechsel des Masters auf LEDE in die Stable zurückportieren können ohne alles instabil zu machen. Also müssten wir auf die übernächste Stable warten.

grmp, 2016-08-16:
If there is a working backport which only needs a check to add support for Gluon, I don't see any reason why it shouldn't be done.
C2600 is a promising hardware and that way we could use and test it upfront.

RubenKelevra, 2017-03-27:
Well, since you now changed to LEDE the whole backport-part is no longer needed. But the build-script-stuff of LEDE is that much changed that I'm not capable of merging the original C2600 patches to LEDE.

It would be great if somebody could just take them and bring them to LEDE, since it looks like that there is still the support for the C2600 missing.

After that, we might give it another try in Gluon? :)

NeoRaider, 2017-03-27:
Test build running, C2600 support has been in OpenWrt/LEDE for a long time.

commit 57070a56 (2017-03-27): ipq806x: add new target and support for TP-Link Archer C2600

Completely untested for now.

Closes #835
closed 2017-03-27 by NeoRaider