freifunk-gluon/gluon #841: Unterstützung für neue Ubiquiti-Geräte AC-Geräte möglich?

RubenKelevra, 2016-08-03:
Hallo zusammen,

der Unifi AP AC Pro wird ja bereits unterstützt, sind die Litebeams AC, Litebeams AC AP sowie die NanoBeams AC denn auch unterstützbar?

Wir finden die Hardware-Plattform sehr interessant als Ablösung von Nanostations m5, die Hardware-Spezifikationen wirken ja ähnlich zu den Ubiquiti Unfi AP AC Pro, wäre also support grundsätzlich möglich?

Danke im Voraus!

NeoRaider, 2016-08-08:
Ich habe eine Rocket AC zur Verfügung und kann mir die demnächst mal anschauen.

RubenKelevra, 2016-08-08:
@NeoRaider das wäre natürlich klasse! Wir haben eine Litebeams AC sowie Litebeams AC AP zum Entwickeln gekauft und würden gern helfen. Was würde dir da weiterhelfen? :)

ArwedSchmidt, 2016-08-26:
+1 weil Gigabit Ethernet und AC.
Einzig: War Airmax da nicht unabschaltbar in Quarz gegossen und das ganze deshalb ein Problem?

RubenKelevra, 2016-08-27:
Also in der Orginal-Firmware kann man es ein und ausschalten. Woher kommt deine Info?

ArwedSchmidt, 2016-08-27:
Es war mal unmöglich ein Clientnetz aufzumachen / diese Option war unauffindbar. Wohlgemerkt reden wir von Airmax **AC**. Das normale Airmax (Nanostation M2 etc) war abschaltbar. Bei welchem Modell in welcher Firmwareversion kannst du Airmax AC abschalten? 

Cheers,
Arwed

RubenKelevra, 2016-08-27:
Ah ich war bei der Litebeam m5 :)

Adorfer, 2016-09-07:
Ich kann gern für die Entwicklung LitebeamAC und LitebeamM5 zur Verfügung stellen. 
Denn selbst wenn das mit dem TPM&WDS nicht funktionieren sollte (wovon ich nicht ausgehe), selbst Indoor würde ich die gern verwenden, weil ich inzwischen Locations habe, wo die Mehrheit der Clients 5GHz kann, ein indoor5GHz-Mesh auch gut wäre. Und für die paar verbliebenenen 2.4only würde ich halt noch einen 841er vorn ins Fenster stellen.

RubenKelevra, 2016-09-07:
> TPM&WDS
@Adorfer meinst du TPC und DFS? :)

NeoRaider, 2016-09-07:
Ich bin gerade an der Rocket AC dran. Wenn die läuft, kann ich mir andere Airmax-AC-Geräte anschauen.

Adorfer, 2016-09-07:
@RubenKelevra Fixed, Danke.

ArwedSchmidt, 2016-09-07:
Ohne DFS ist doch 5 GHz im Außenbereich mit halber Leistung (bis 500mW) auch zulässig, oder interpretier ich da was falsch? Wäre also in jedem Fall ein Gewinn mit Gluon. Gerade die Litebeam AP Variante würde zum Nachfolger der Nanostation werden

RubenKelevra, 2016-09-07:
@ArwedSchmidt ohne TPC aber mit DFS darf man 500 mW , ohne DFS nur 200 mW im Channel 100+

Alle Kanälte unter 100 darf man noch weniger, da maximal 200mW weiß grad nicht genau wie man da noch weiter reduzieren muss.

MPW1412, 2016-09-22:
Ich habe mich damit kürzlich beschäftigt, weil ich abseits von Freifunk mit einer PowerBeam AC für mich privat das Uni-WLAN auf eine größere Entfernung anzapfen wollten.

Erkenntnis dabei, auch so bestätigt durch Ubiquity: Alle Richtfunkgeräte mit „AC“ im Namen können kein normales 802.11n. Zumindest mit der Originalfirmware, weil dieser spezielle Ubiquity-Modus hardwaremäßig implementiert ist. Ubiquity plant nicht, 802.11n zu implementieren. Ob das seitens OpenWRT bzw. mit anderen Treibern überhaupt möglich ist, weiß ich nicht.

belzebub40k, 2016-09-22:
@MPW1412 Versuch doch mal ob es mit der airOS8 Beta funktioniert. In dieser wurde die airMAX N und airMAX AC Geräte kompatibel gemacht. Möglicherweise funktioniert damit auch die Verbindung zu 802.11n Geräten die nicht von Ubiquiti stammen.

https://community.ubnt.com/t5/airMAX-AC/airOS-7-airMax-M-802-11n-compatibility/td-p/1162028

MPW1412, 2016-09-22:
Ich kann mich irgendwie gerade nicht für die Beta von Ubiquity freischalten, weil die Webseite hängt. Ich probiere es später nochmal.

Ich glaube aber nicht, dass es funktioniert. Nur weil sie die beiden AirMax-Modi kompatibel gemacht habne, heißt das noch lange nicht, dass die Geräte jetzt als normaler AccessPoint oder als Station (Client-Modus) funktioniert.

ArwedSchmidt, 2016-10-12:
@NeoRaider Konntest du bereits ein Gefühl entwickeln, ob es einen Gluon-Support für die AC-Geräte wird geben können? Es geht nur darum, ob es längerfristig sinnvoll ist solche Geräte zu verbauen oder man sagen kann es wird auf Ewig Stockfirmware bleiben auf der HW. Wenn ich das richtig sehe gibt es seitens OpenWrt keinen Support für die Dinger.

Cheers aus Köln,
Arwed

rotanid, 2017-02-12:
related to #1021 - although this issue here is more a discussion and has a very general subject

RubenKelevra, 2017-04-04:
Well, it seems that Ubiquiti currently enabled a non-Airmax mode in their latest beta firmware. Just a report from our community, not tested myself, thought.

rotanid, 2018-01-19:
closing this issue as it is very vague/unspecific and should be handled at upstream LEDE/OpenWrt.
if there is support in LEDE/OpenWrt for any device that you want to use in Gluon, please open a specific issue for that device - or even better create a pull request on your own, that's the much faster way.

closed 2018-01-19 by rotanid