freifunk-gluon/gluon #842: problem with upgrade from 2016.1.5 to master, duplicate wifi mac

rotanid, 2016-08-03:
tonight our routers on our experimental firmware branch updated from an 2016.1.5-based firmware to an master-based one.
all of these routers have both ibss and 11s meshing enabled.

in one location, two ath9k dual-band routers (WDR3600) mesh over 2,4ghz (too far away for 5ghz to work).
the one which has the vpn-uplink also has private wifi enabled, so 4 wifis on 2 radios makes 8.

it broke a bit due to the update:
the client_radio0 mac address wasn't similar to the client_radio0 as it is on other 2016.1.5-based devices.
it was the same as the one for wlan0-2 and as a consequence, hostapd didn't want to start and the 2,4ghz radio wifi interfaces were down.

i could fix it by copying the mac from client_radio1 in /etc/config/wireless and only change the 3rd octet (compared the macs on other devices to come up with that solution)

what could have caused that issue, are 8 wifi interfaces too much for master branch upgrade scripts?

RubenKelevra, 2016-08-07:
Would you be so kind to check the wan-mac for gigabit-devices? I think we've discovered a but there too. On our experimental dev build with the old building-system all wan-macs on gigabit-devices were random.

rotanid, 2016-08-08:
what exactly is "experimental dev build with the old building-system" ? :-D
if you talk about the mac of the "br-wan" interface, it is the same as the mac of the "client1" interface on the two wdr3600 i am able to check at the moment. @Brother-Lal might have another wdr3600 running the same gluon-master-based firmware

NeoRaider, 2016-08-08:
I'm a bit confused why client0 and wlan0-2 have the same MAC address... client0 is expected to have a completely different address than in v2016.1.x, but I don't see how wlan0-2 gets the same address.

Is there any address set for the private-wifi interface in `/etc/config/wireless`?

rotanid, 2016-08-08:
no, wan_radio0 and wan_radio1 have no mac address set in that config file

rotanid, 2016-08-08:
no, wan_radio0 and wan_radio1 have no mac address set in that config file.
i have a few of the wdr3600 left, i could do the upgrade again but i don't think that it will be different...

benerudolph, 2016-08-15:
I do experience the same behavior stated above on an Archer C7 (it has two wifi interfaces) under the very same circumstances.
My logread prints this, when the system starts the hostapd process that is supposed to manage the 
WiFi interfaces:
```
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Configuration file: /var/run/hostapd-phy0.conf
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Duplicate BSSID e8:de:27:xx:yy:zz on interface 'wlan0-2' and 'client0'.
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Failed to set up interface with /var/run/hostapd-phy0.conf
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Failed to initialize interface
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Command failed: Invalid argument
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio0 (1402): Device setup failed: HOSTAPD_START_FAILED
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio1 (1403): Configuration file: /var/run/hostapd-phy1.conf
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio1 (1403): Duplicate BSSID e8:de:27:xx:yy:zz on interface 'wlan1-2' and 'client1'.
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio1 (1403): 1 errors found in configuration file '/var/run/hostapd-phy1.conf'
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio1 (1403): Failed to set up interface with /var/run/hostapd-phy1.conf
Sat Jul 30 14:19:17 2016 daemon.notice netifd: radio1 (1403): Failed to initialize interface
Sat Jul 30 14:19:18 2016 daemon.notice netifd: Interface 'wan' is now up
Sat Jul 30 14:19:18 2016 daemon.notice netifd: radio1 (1403): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Sat Jul 30 14:19:18 2016 daemon.notice netifd: radio1 (1403): Command failed: Invalid argument
Sat Jul 30 14:19:18 2016 daemon.notice netifd: radio1 (1403): Device setup failed: HOSTAPD_START_FAILED
```

commit aaa5ca3a (2016-08-30): Reorganize MAC addresses, always explicitly set address for private WLAN

Fixes #842
commit 6cddaedf (2016-09-07): Reorganize MAC addresses, always explicitly set address for private WLAN

Fixes #842
closed 2016-09-07 by NeoRaider