freifunk-gluon/gluon #845: error: private WLAN ssid with more than 32 characters

rubo77, 2016-08-05:
If you choose to share your private WLAN, and choose an SSID with more than 32 characters, wifi will not work any more.

the Router boots fine, but in logread` you see:

```
Device setup failed: HOSTAPD_START_FAILED
shake to <mesh_vpn_backbone_peer_ffki_vpn0>[89.27.152.8:11235]...

 2016 daemon.notice netifd: radio0 (4471): Configuration file: /va

Line 27: invalid SSID 'macht mit bei Freifunk! ist ganz einfach!'
1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Failed to set up interface with /var/run/hostapd-phy0.conf
Failed to initialize interface
cat: can't open '/var/run/wifi-phy0.pid': No such file or director
Command failed: Invalid argument
Device setup failed: HOSTAPD_START_FAILED
Configuration file: /var/run/hostapd-phy1.conf
Line 28: invalid SSID 'macht mit bei Freifunk! ist ganz einfach!'
1 errors found in configuration file '/var/run/hostapd-phy1.conf'
Failed to set up interface with /var/run/hostapd-phy1.conf
Failed to initialize interface
```

rubo77, 2016-08-05:
workaround on the console:

change the ssid:

    uci set wireless.wan_radio0.ssid='Freifunk! Ist ganz einfach!'
    uci set wireless.wan_radio1.ssid='Freifunk! Ist ganz einfach!'

kb-light, 2016-08-07:
out of [wikipedia.org](https://en.wikipedia.org/wiki/Service_set_(802.11_network)#Service_set_identification_.28SSID.29): 

> Each BSS or ESS is identified by a service set identifier (SSID) - a series of 0 to 32 octets.

so the max length of a SSID is 32 characters.

rubo77, 2016-08-07:
that is the problem: in config mode, you can enter more than 32 chars and this is stored in the uci settings, which will result in a broken state of the router

kb-light, 2016-08-07:
@rubo77: please test if #848 solves this issue

PR #848 (2016-08-08): gluon-luci-private-wifi: limit ssid to 32 characters
fixes #845
commit 2789aebc (2016-08-08): gluon-luci-private-wifi: limit ssid to 32 characters (#848)

Fixes #845
closed 2016-08-08 by NeoRaider