freifunk-gluon/gluon #851: The correct question to opt-in for an node-offline-email-alert

rubo77, 2016-08-07:
We would like to add an option like this in config mode:

[x] I would like to be notified, if my node is offline
> ... tell us how to reach you by the contact field or a private email or an online nodewatcher system, that will be available soon

Es gibt da [verschiedene Meinungen](https://github.com/freifunk-gluon/gluon/pull/843#issuecomment-237911727), was die rechtlich korrekte Formulierung bzw. Implementierung wäre, wie man den Usern ermöglichen könnte bei so einem Alert-Dienst mitzumachen.

Meine Vorstellung wäre, die Formulierung im Config mode in der site-conf definierbar zu machen um zu ermöglichen auf vorhandene Dienste, oder Vorgehensweisen zu verlinken.

Die simpleste Vorgehensweise wäre, dass man mit dieser Erlaubnis aus dem alfred-string dann z.B. als Keys-Verwalter, der ja die initiale email mit dem vpn-key erhalten hat, dadurch die Erlaubnis bekommt den Knotenbetreiber anzuschreiben wenn sein Knoten offline ist.

RubenKelevra, 2016-09-09:
Am einfachsten ist, den Usern nachdem ein neuer Node auftaucht eine Mail zu schicken, wenn sie es an haben wollen sollen Sie auf einen Link klicken. Wenn man an jede Mail einen Opt-Out Link hängt sollte das rechtlich Safe sein. :)

jplitza, 2016-09-11:
Das ist halt ne Mail mehr und verhindert auch keinen Missbrauch (ich kann mir einfach ständig neue virtuelle Nodes erzeugen und damit rumspammen). Am sinnvollsten wäre es, einfach Mails zu verschicken mit nem Opt-Out drunter, oder nur die erste Mail zu schicken und dann ein Opt-In zu fordern - aber halt erst wenn der Knoten offline geht. Außerdem sollten Opt-Ins denke ich pro E-Mailadresse, nicht pro Knoten, gespeichert werden.

RubenKelevra, 2016-09-13:
@jplitza in dem Ticket gehts nicht um "missbrauchsverhinderung" sondern darum was man rechtlich tun muss. Und rechtlich muss man eine Mail schicken, mit der man prüft ob der Empfänger auch derjenige ist, der die Mailadresse eingetragen hat.

jplitza, 2016-09-22:
Okay, da es ums rechtliche geht sollte hier denke ich auf den ["Rechtslage"-Abschnitt zum Opt-In bei Wikipedia](https://de.wikipedia.org/wiki/Opt-in#Deutschland) hingewiesen werden. Selbst falls (!) unsere Node-Offline-Benachrichtigungen als Werbung gelten (was ich schon bezweifle), sind wir nicht zum Double-Opt-In verpflichtet.

RubenKelevra, 2017-03-27:
Und man sollte natürlich auf dem Server eine Liste pflegen wo die ausgehenden Mail-Adressen für Anfragen gehasht zwischengespeichert werden. Damit nicht jemand ein paar hundert Mails über den Server rausschickt.

Das ist aber alles eigentlich kein Gluon-Stuff höchstens ne Checkbox könnte man in Gluon einbauen, ob man das will oder nicht. :)

rotanid, 2018-01-19:
Closing. if there is still need for discussion, this issue can be used although it is closed.
But using the mailing list or the Forum might be better for discussing this.

closed 2018-01-19 by rotanid