freifunk-gluon/gluon #852: kern.warn kernel: br-client receives packets on bat0 with own address as source address

Byggvir, 2016-08-10:
Hello 

I get lots of kernel warnings (5-6) per second about packets on bat0 with own address as source address. 

Thu Aug  4 11:15:12 2016 kern.warn kernel: [281419.700000] br-client: received packet on bat0 with own address as source address

My last image generated on 22nd April does not show these kernel warnings. 
Only the new images generated on 22nd July and 31st Juli (branch v2016.1.x) show these warnings.

Routers TPLINK TL WR841 ND, 1043 and CPE 210.

Thanks 

Thomas

jplitza, 2016-08-10:
Is this the same issue as #834? I.e. do you have Bridge Loop Avoidance activated?

Byggvir, 2016-08-10:
Yes, it's the same error message.
Sorry I only scanned the titles for duplicates. Didn't look in the issues.

closed 2016-08-10 by NeoRaider