freifunk-gluon/gluon #853: Can't access config interface after assigning all switch ports to private LAN

JohannesRudolph, 2016-08-11:
I run Freifunk Darmstadt's gluon derivate on an Archer C7. I have assigned all 4 switch ports to the private LAN network via ssh (see https://wiki.openwrt.org/doc/uci/network/switch). Unfortunately I'm now no longer able to access the web interface when putting the router in config mode. DHCP appears defunct and I can't access it on 192.168.1.1 with manual IP configuration either. Accessing the config web interface from the WAN interface doesn't work either

I'm not sure how gluon works, but if it uses openwrt safe mode it should also load a default switch config instead of the user defined one. 

Any idea how I can "unbrick" this router? I will need to ssh into it I suppose while it's running, any tips appreciated.

NeoRaider, 2016-08-11:
The setup mode removes all interface configurations from the network config, leaving only the switch config. Then a "setup" interface is added, using the interface name set in `/lib/gluon/core/sysconfig/setup_ifname`. You should be able to fix it by replacing the interface name in that file.

JohannesRudolph, 2016-08-12:
Fantastic, I was able to ssh into the node and fix it.

closed 2016-08-12 by JohannesRudolph