freifunk-gluon/gluon #856: Support für TL-WR810N oder TL-WR802N

florianjb, 2016-08-14:
Servus zusammen, ist schon mal versucht worden, Freifunk-FW auf einem TL-WR810N oder TL-WR802N zum laufen zu bringen?

Casey1979, 2016-08-14:
Es gibt folgenden Patch um OpenWRT für den TL-WR802N bauen zu können:
https://patchwork.ozlabs.org/patch/571269/

Von dort kann man dann eine Gluon-Firmware ab 2016.1.4 für den TL-WR841N-V9 installieren die auch läuft, aber nicht konfigurierbar ist weil das Gerät mit der jetzigen Firmware nur einen WanPort hat über den man Gluon aber nicht konfigurieren kann.
Deshalb muss nun das Gerät geöffnet werden, ein Widerstand herausgebrochen werden um auf der seriellen Konsole senden zu können und dann der zweite Ethernetport br-config hinzugefügt werden, danach ist Gluon konfigurierbar und mesht nach dem Speichern ganz normal.
Nächstes Problem, nach dem aktivieren der seriellen Konsole hat sich bei mir der Stromverbrauch um 50% erhöht, von 220mA auf 330mA, das Gerät wird dann aufgrund seiner sehr geringen Größe sehr heiss und hat bei Nutzung des Ethernetports sogar so eine Hitze entwickelt das der Wlanteil bis zum Reboot ausgestiegen ist.
Nach aufkleben eines kleinen Kühlkörpers für den RaspberryPi ist es ein super Mobilnode der einem normalen TL-WR841N trotz der winzigen Anternnen nur um wenige dB nach steht.
Das Gerät wird aber immer noch so heiss das es nicht ohne Aufsicht betrieben werden sollte, auch wenn es selbst im Hochsommer nicht mehr ausgestiegen ist.

Wenn jemand den Patch in OpenWRT oder LEDE integriert dürfte einer Unterstützung in Gluon nichts im Weg stehen, dann sind hoffentlich auch die anderen Probleme ein Thema der Vergangenheit.

Für diejenigen die den Weg gehen wollen ein Tipp zum Gehäuse öffnen. Mit einem dünnen und spitzen Taschenmesser über dem MiniUSB-Port vorsichtig zwischen die Gehäuseteile stechen und die Oberseite einfach mit Gewalt heraushebeln, es sollte mit ein bischen Feingefühl nichts dabei kaputt gehen. Der Widerstand ist der gleiche wie beim TL-WR841N-V9 ist aber der Hitzeentwicklung wegen nicht zu empfehlen.

Würde mich auch drüber freuen wenn es eine offizielle Unterstützung geben würde ...

NeoRaider, 2016-08-15:
TL-WR810N wird von OpenWrt/LEDE unterstützt, ich kümmere mich beizeiten mal um nen Backport.

Den Patch für den TL-WR802N müsste mal wer aufräumen und ordentlich submitten bei LEDE.

Casey1979, 2016-08-30:
Du meinst so wie dieser Patch vom Mai? https://patchwork.ozlabs.org/patch/617629/

NeoRaider, 2016-08-30:
Ja, das sieht deutlich besser aus.

Casey1979, 2016-10-30:
Angesichts des fehlenden Fortschritts bei der Patchintegration in OpenWRT (der Patch liegt dort seit Mai auf Halde) der bei Lede vermutlich nicht anders ausfallen würde, da es ja die gleichen beteiligten sind, macht es vielleicht mehr Sinn nach dem Wechsel der Basis den GL-AR300M zu unterstützen der in LEDE schon integriert worden ist, sprich den Aufwand auf ein Minimum beschränkt.
Zumal der Router für etwas mehr Geld in jeder Hinsicht mehr bietet und in Deutschland sowohl mit internen als auch externen Antennen erhältlich ist.

NeoRaider, 2016-11-01:
OpenWrt benutzt das Patchwork nicht mehr, die sind vollständig auf Github gewechselt.

Ansonsten ist mir aufgefallen, dass der Patch völlig kaputt ist (Whitespace-Probleme, die das Patchfile ungültig machen), ansonsten würde ich den einfach mal nach LEDE pushen...

Casey1979, 2016-11-21:
Hab nun mal den GL-AR300M ausprobiert, das Gluon LEDE NOR Image läuft darauf, allerdings im Moment noch mit folgender Einschränkung.
Der Router hat zwei Firmwares gleichzeitig im Flash, eine im NOR und eine im NAND Speicher die bevorzugt gestartet wird. Wobei man die NOR version nur aus dem laufenden NAND Image heraus händisch starten kann. Die Kommandos aus rc.local heraus abzusetzen hat warum auch immer nicht zum Erfolg geführt.
NAND Support für den GL-AR300M soll für Lede jedoch nochmals eingereicht werden, erst danach wird es Sinn machen. Patches für OpenWRT existieren zwar dort https://github.com/domino-team/openwrt-cc aber für Gluon nicht im passenden Format und vielleicht auch noch mit anderen Problemen.
https://www.gl-inet.com/forums/topic/gl-ar300m-with-lede-trunk/
https://github.com/lede-project/source/pull/272

rotanid, 2017-11-12:
according to the following upstream commits, these devices are both supported by LEDE for a long time:
https://github.com/lede-project/source/commit/75629fb2a86cae1cb3829216215c843d93d0c3b7
https://github.com/lede-project/source/commit/a71a8955f20e3c3ac13323072a70362f3c9d80fd
adding support to Gluon should be very easy therefore.

rotanid, 2018-01-19:
@florianjb i've added support for these 2 devices:
708aae501578989248f79efd18965334c0ffbf92
c4e3cca1d3e634c8a4cf83dfb4998a86d5bd318a
these are marked as BROKEN for now, if you report them as working, we may remove the BROKEN flag.

closed 2018-01-19 by rotanid

Marssl78, 2018-10-13:
Hello, there!
I did the Testing for the TL-WR810N v1.1(EU) with two Firmwares I built by myself (Gluon Rel. v2018.1.1 and Master for checking the sysupgrade functionality).

     * [x ]  must be flashable from vendor firmware
       
       * [x]  webinterface
       * [x]  tftp
 
     * [x]  must support upgrade mechanism
       
       * [x]  must have working sysupgrade
         
         * [x]  must keep/forget configuration (if applicable)
           _think `sysupgrade [-n]` or `firstboot`_
       * [x]  must have working autoupdate
         _usually means: gluon profile name must match image name_
 
     * wired network
       
       * [x]  should support all network ports on the device
       * [x]  must have correct port assignment (WAN/LAN)
 
     * wifi (if applicable)
       
       * [x]  association with AP must be possible on all radios
       * [x]  association with 802.11s mesh must be working on all radios
       * [x]  ap/mesh mode must work in parallel on all radios
 
     * led mapping
       
       * power/sys led
         
         * [x]  lit while the device is on
         * [x]  should display config mode blink sequence
           (https://gluon.readthedocs.io/en/latest/features/configmode.html)
       * radio leds
         
         * [ ]  should map to their respective radio
         * [ ]  should show activity
       * switchport leds
         
         * [ ]  should map to their respective port (or switch, if only one led present)
         * [ ]  should show link state and activity [not possible for this device]
 
     * [x]  reset/wps button must return device into config mode
 
     * [x]  primary mac should match address on device label (or packaging) (https://gluon.readthedocs.io/en/latest/dev/hardware.html#notes)

As there are no Switch / Radio-LEDs, for my understanding testing has successfully passed :-)

blocktrron, 2018-10-14:
@Marssl78 can you paste the raw output of `lua -e 'print(require("platform_info").get_image_name())'` ?
Looks good otherwise.

Marssl78, 2018-10-15:
This gives me:

tp-link-tl-wr810n-v1

rubo77, 2018-10-15:
Once it works, you can gemove the BROKEN flag here too: https://openwrt.org/toh/hwdata/tp-link/tp-link_tl-wr810n_v1.1_eu

Marssl78, 2018-10-22:
Can I give further Assistance here?

Dark4MD, 2018-10-23:
Beim Freifunk Hannover gibt es für den WR810N v1 ein Stable Image, das hab ich sogar schon eingesetzt problemlos aber noch nicht lange. Kann den gerne morgen mal wieder Anstecken und eine Weile laufen lassen.
Ich hab auch noch einen WR802N v2 auf dem ich bereits ein Master drauf laufen lassen hab, lief auch soweit problemlos. Kann ich gerne auch mal alles durch testen bei Bedarf.

blocktrron, 2018-10-24:
@Marssl78 

> the profile name has no HW Version (v1) in it

What is meant with that? Are you referring to the devices profile name in OpenWRT? For the Autoupdater it is relevant that the filename device-part matches the string given by `lua -e 'print(require("platform_info").get_image_name())'`, so it looks good to me.

Just wanted to double-check this point as @rotanid pointet out this phrasing is a bit ambiguous.

Marssl78, 2018-10-24:
I was referring to Rotanid's 

[708aae5](https://github.com/freifunk-gluon/gluon/commit/708aae501578989248f79efd18965334c0ffbf92)

where, in line 164 the device is mentioned twice without any Hardware-Version. But the image I got from the build-process in fact does have a "-v1" in the filename.

blocktrron, 2018-10-24:
The output name got changed in this commit: https://github.com/freifunk-gluon/gluon/commit/210d97c53eb147ad9519118634c71e28eace4694

So everything seems to be named as expected.

rotanid, 2018-10-24:
ok, merged your PR removing the BROKEN flag

PR #1557 (2018-10-24): ar71xx: remove BROKEN flag for TL-WR810N v1
This removes the BROKEN flag for the TP-Link TL-WR810N v1 as it is verified to be working by @Marssl78

Fixes #856
commit c1967fe3 (2018-10-24): ar71xx: remove BROKEN flag for TL-WR810N v1

Fixes #856