freifunk-gluon/gluon #860: Archer C7v2 - Keine Installation möglich - Error: 18005

zawhie, 2016-08-22:
Ausgangslage: Archer C7 Version 2, Firmware Version:	3.15.1 Build 160616 Rel.44182n

Versuche ich nun die gluon-remscheid-4.2.0-tp-link-archer-c7-v2.bin übers Webinterface zu installieren wird der Vorgang mit der folgenden Nachricht abgebrochen:

_Error Code: 18005 
Upgrade unsuccessfully because the version of the upgraded file was incorrect. Please check file name._ 

Auf der Seite von TP-Link (http://www.tp-link.de/download/Archer-C7_V2.html#Firmware) steht zur aktuellen Firmware:
_2.The EU firmware was specialized for CE certification and can't be downgraded to other version, please click here for choosing your region and selecting the most suitable firmware version to upgrade._

NeoRaider, 2016-08-22:
TP-Link added a new region identification to its firmwares, I'll see if I can add support for this to OpenWrt/Gluon.

closed 2016-08-28 by NeoRaider

RubenKelevra, 2016-09-07:
@zawhie thanks for your ticket, I'm one of the dev team for the firmware for Remscheid. The 4.2.1 will contain this upstream-fix and is going to be released at the weekend.

RubenKelevra, 2016-09-07:
@neoraider the milestone of this ticket is wrong, not sure if you care :)

commit 639c12b8 (2016-08-28): ar71xx-generic: add support for new TP-Link region codes

Generate flashable images for the Archer C7 v2 with current stock firmware
again.

To set the region code, the GLUON_REGION variable must be set to "us" or
"eu" in site.mk or as a make argument.

Fixes #860
commit adff3dae (2016-08-28): ar71xx-generic: add support for new TP-Link region codes

Generate flashable images for the Archer C7 v2 with current stock firmware
again.

To set the region code, the GLUON_REGION variable must be set to "us" or
"eu" in site.mk or as a make argument.

Fixes #860