freifunk-gluon/gluon #861: change gateway if no internet connection

rubo77, 2016-08-23:
The nodes should check, if they can reach an internet address through the gateway, and if not, automatically drop the gateway and try connecting to another gateway

Adorfer, 2016-08-23:
A gateway should monitor itself constantly if it can reach the internet via its route for Batman clients. (table 42 or whatever). 
and if this fails, step down from announcing gw.

NeoRaider, 2016-08-23:
The "gateway" flag announced by the gateways is a signal that the nodes should understand as "use me as default route". If a gateway is wrongfully announcing itself, fix your gateway.

closed 2016-08-23 by NeoRaider

rubo77, 2016-08-23:
What about nodes, that use a gateway and then the gateway stops announcing as gateway, do the nodes notice this fast and search for another gateway?

NeoRaider, 2016-08-23:
Yes. Clients will get the new gateway the next time they renew their DHCP lease.

rubo77, 2016-08-23:
So, if you use this method, if the gateway stops transmitting the gateway status, the router will be one hour without internet connection!

If we would add a check in the nodes, if it can resolve addresses through the chosen gateway, and if not, choose another gateway (earlyer DHCP Lease), we would enhance the stability and gateway redundance.

NeoRaider, 2016-08-23:
1. The Gluon nodes don't use the gateway feature at all for their internet connection, as the nodes only have IPv6 and the gateway feature only affects IPv4.
2. The node can't make the clients choose another gateway. Make your leases as short as possible (we use 5 minutes in Lübeck, shorter leases aren't supported by all clients). Please read up how batman-adv's gateway selection works.

Some of these limitations will be irrelevant when we move the gateway address to the nodes themselves, but moving towards new network topologies is another subject.