freifunk-gluon/gluon #867: Keine Verbindung, wenn 5 GHz deaktiviert und micrond aktiv

zawhie, 2016-08-29:
Firmware: gluon-remscheid-4.2.0-tp-link-archer-c7-v2.bin
Access-Point: TP-Link Archer C7v2

Problem: Sobald das Freifunk auf 5 GHz deaktiviert wird, verlieren die Clients auf 2,4 GHz die Verbindung, unabhängig, ob diese mit Freifunk oder dem privaten WLAN (2,4 GHz und 5 GHz) verbunden sind. Wird Freifunk auf 5 GHz wieder aktiviert läuft alles ohne Probleme.

(Freifunk 2,4 GHz: an, Freifunk 5 GHz: aus, Private 2,4 GHz: aus, Private 5 GHz: an => Verbindung wird verloren
Freifunk 2,4 GHz: an, Freifunk 5 GHz: an, Private 2,4 GHz: aus, Private 5 GHz: an => Stabil)

work-around: Nachdem micrond gestoppt wurde bleibt die Verbindung stabil, auch wenn Freifunk auf 5GHz deaktiviert wurde.
(Freifunk 2,4 GHz: an, Freifunk 5 GHz: aus, Private 2,4 GHz: aus, Private 5 GHz: an, _micrond gestoppt_ => stabil)

ecsv, 2016-08-29:
Ausm IRC

    <ecsv> eihwaz: das sieht mehr aus als wuerde bei euch irgend ein script laufen welches es nicht mag wen 5ghz deaktiviert ist
    <ecsv> micrond == cron
    <ecsv> also daher hoechstwahrscheinlich nur in eurer firmware und nicht in gluon

Wahrscheinlich ist es irgend ein ath9k-Workaround-Skript welches von micrond aller x-minuten getriggered wird.

NeoRaider, 2016-08-29:
Das Problem liegt höchstwahrscheinlich im Offline-SSID-Changer der NRW-Firmware, auf jeden Fall ist es kein Script von Gluon selbst.

closed 2016-08-29 by NeoRaider

RubenKelevra, 2016-08-29:
Der Fehler ist bei uns schon bekannt und wird demnächst behoben. Nur als Referenz:

https://github.com/VfN-NRW/gluon-firmware-builder/issues/166