freifunk-gluon/gluon #868: error while compiling the kernel

jjsarton, 2016-08-29:
I have tried to build a target under fedora 24 which use gcc-Version 6.1.1 20160621 (Red Hat 6.1.1-3).
openwrt/build/<GLUON_TARGET>/openwrt/build_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04/gcc/cp/cfns.h must be modified:
54: #ifdef __GNUC__
55: __inline
56: #endif
57: const char * libc_name_p (const char *, unsigned int);
produce an error deleting / commenting in the line 54 to 56 resolve the problem.

A file .openwrt/build_dir/host/u-boot-2014.10/include/linux/compiler-gcc6.h must also be created, I copied simply compiler-gcc5.h to compiler-gcc6.h

The last problem I have not fixed, is whitin openwrt/build_dir/host/node-v0.12.7/out
 
Traceback (most recent call last):
  File "../../tools/js2c.py", line 37, in <module>
    import bz2
ImportError: No module named bz2

jjsarton, 2016-08-29:
the node binary produced for the tool chain is also wrong I get:
make[4]: Entering directory '/.../gluon/package/gluon-status-page'
cd /.../gluon/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/gluon-status-page && node r.js -o build.js && node r.js -o cssIn=css/main.css out=style.css && /.../gluon/build/ar71xx-generic/openwrt/staging_dir/host/bin/m4 index.html.m4 > index.html
bash: line 1: 13143 Segmentation fault      (core dumped) node r.js -o build.js

If I use node for my system (v4.5.0) there is no problem.

NeoRaider, 2016-08-29:
All of this sounds like old bugs already fixed in Gluon master and v2016.1.x.

jjsarton, 2016-08-30:
Ich habe die master branch geklont, es sollte eigentlich funktionieren.  
Möglicherweise fehlte die Eingabe des kommando "make update"
Ich habe die Release 2016.1.5 installiert und konnte nun alles erzeugen.

closed 2016-08-30 by NeoRaider