freifunk-gluon/gluon #873: Make the label for altitude definable in the site folder

rubo77, 2016-08-31:
We need to remove the current strings and replace them with a placeholder.

rubo77, 2016-09-12:
the setting should be completely self definable in the i18n-files in the site-conf, so you can for example also label it as "Storey" or "Height above Streetlevel"

Adorfer, 2016-09-12:
When asking for "height" we have to take into account, that especially in former eastern germany many "public visible" waymarks (bridges, floods-levels) are still in SNN76 and not transformed to DHNN92. 
(And some people prefererr the GPS-WGS84 geoid.)

rubo77, 2016-09-12:
Thats why this Issue asks for a completely user definable text for the altitude section for every community.

You should define the exact wording in your community site-repository, so you can decide for yourself, what the users should enter in this field.

Adorfer, 2016-09-12:
the community in Aachen for example is challenged by the fact that height grid in belgium is more than 2m off from the german grid.

Adorfer, 2016-09-12:
P.S. Vor conversion you may link in your firmware: 

http://jules.unavco.org/Geoid/Geoid

Example output: 
Your Input Coordinates and GPS Height:

> Latitude = 50.2477777777778° N = 50° 14' 52" N
> Longitude = 8.95416666666667° E = 8° 57' 15" E
> GPS ellipsoidal height = 15 (meters)
> Geoid height = 48.235 (meters)
> Orthometric height (height above EGM96 geoid which approximates mean sea level) = -33.235 (meters)
>

closed 2016-09-13 by rubo77

rubo77, 2016-11-26:
As described in the followup: 

You can already achieve this by addint this to your site config in `i18n/de.po`:

```
msgid "Altitude"
msgstr "Höhe (3m pro Stock)"

msgid ""
"Specifying the altitude is optional and should only be done if a proper "
"value is known."
msgstr ""
"Die Höhenangabe ist optional und kann geschätzt werden mit 3m pro Stockwerk."
```

PR #883 (2016-11-30): gluon-config-mode-geo-location: Make the label for altitude definable…
… in the site folder

closes #873