freifunk-gluon/gluon #874: br-client ipv6 address

Byggvir, 2016-08-31:
After a firmware update I get only one inet6 address on the br-client Scope:Link

br-client Link encap:Ethernet  HWaddr 30:B5:C2:A6:1B:A0  
          inet6 addr: fe80::32b5:c2ff:fea6:1ba0/64 Scope:Link

instead of two addresses with Scope: Global an Scope:Link

br-client Link encap:Ethernet  HWaddr C4:E9:84:F9:CB:AE  
          inet6 addr: fda0:747e:ab29:2241:c6e9:84ff:fef9:cbae/64 Scope:Global
          inet6 addr: fe80::c6e9:84ff:fef9:cbae/64 Scope:Link

This means that the routers can't be reached via ping6 and ssh anymore.

viisauksena, 2016-08-31:
inside batman network your router should still be reachable through link local v6 adresesses by explicit adding outgoing if  ... like this 
ssh -lroot  fe80::32b5:c2ff:fea6:1ba0%wlan0

T-X, 2016-08-31:
Is the global IPv6 address coming from a gateway? Which Gluon version did you have before and after?

Byggvir, 2016-09-01:
Hello,

this is a little bit strange. It took some hours to get / set the address with scope global. This morning - after 6 to 8 hours uptime - the three updated routers didn't have a inet6 address Scope: global. Now - 10 hours later - two routers have a global address. I didn't reboot them in the meantime. The third is unreachable, but that is normal for this CPE210 - CPE210 link over 160m.

I rebooted two routers, CPE 210 and WR842ND, and they both didn't get or set a scope global address.

The second CPE210 looses connection to the first very often. It was restarted 40 Minutes ago and has no global inet6 address yet.

1. CPE 210 https://map.freifunk-rhein-sieg.de/rheinbach/#!v:m;n:60e327cd5c0c
2. CPE 210 https://map.freifunk-rhein-sieg.de/rheinbach/#!v:m;n:f4f26d6d4250

My second last image was from 15th April, the last from 22nd Juli and now it's is form 6th August.

v2016.1-stable-20160415 / gluon-v2016.1-91-g271e864
v2016.1-stable-20160722 / gluon-v2016.1-208-g1c42aa0
v2016.1-stable-20160806 / gluon-v2016.1-235-gd445cb7

With the previous image I saw this on some - not all - routers and switched them back to the April images.

Byggvir, 2016-09-01:
ssh as  @viisauksena works thanks for the hint!

NeoRaider, 2016-09-01:
The global address is assigned when the node receives a router advertisement from the mesh (usually originating from a gateway server/supernode).

Does downgrading the image fix the issue reliably? I wonder if this has to do something with #688... @T-X

T-X, 2016-09-02:
@NeoRaider: Had been wondering about the same. But looks like an image from the 6th August definitely does not have this recent change yet?

@Byggvir: Does your image include the gluon-ebtables-segment-mld package though?

Byggvir, 2016-09-03:
1. The images doesn't include gluon-ebtables-segment-mld. Never heard of Is this a new package?
gluon-mesh-batman-adv-15 \
gluon-alfred \
gluon-autoupdater \
gluon-config-mode-autoupdater \
gluon-config-mode-contact-info \
gluon-config-mode-core \
gluon-config-mode-geo-location \
gluon-config-mode-hostname \
gluon-config-mode-mesh-vpn \
gluon-ebtables-filter-multicast \
gluon-ebtables-filter-ra-dhcp \
gluon-luci-admin \
gluon-luci-autoupdater \
gluon-luci-mesh-vpn-fastd \
gluon-luci-node-role \
gluon-luci-portconfig \
gluon-luci-private-wifi \
gluon-luci-wifi-config \
gluon-mesh-vpn-fastd \
gluon-neighbour-info \
gluon-next-node \
gluon-radvd \
gluon-respondd \
gluon-setup-mode \
gluon-status-page \
gluon-wan-dnsmasq \
haveged \
iptables \
iwinfo

2. The July image doesn't show this behaviour. (BTW: On my wdr3600 I get only the 5Ghz Channel.)

3. Restarted the Router, but they didn't get a inet6 global address this time.

T-X, 2016-09-03:
@Byggvir, yes gluon-ebtables-segment-mld is a new package recently added to master. But if it's not installed on your side, okay then.

Would it be possible for you to bisect through the Gluon commits between your July and August images? Maybe start with one commit before 7565839fc4761c65a153abae10f6a0370bd65751 (even though you don't have the new package, firewall rules were nevertheless restructured).

Byggvir, 2016-09-03:
I switch from OpenSuSE to Debian for generating the images. 

Tried to generate the images from scratch. Seems that something went wrong when creating the images under Debian. Some lib where now missing. Unfortunately the log file reside on /tmp and are gone.

No need to thing about this problem in the moment. When the new images is workung, there will no need to investigate this issue further.

Byggvir, 2016-09-04:
I generated a new image and updated some test routers wr841, wr842, wr104, wdr3600 and CPE210.
Everything works fine. Even the 3600 I mentioned above starts with 2.4 and 5 GHz.

I think the reason for this behaviour was a mixup of files complied under OpenSuSe and the newer Debian.

closed 2016-09-04 by Byggvir