freifunk-gluon/gluon #875: skip config mode in combination with mesh-on-lan enabled breakes eth0

MPW1412, 2016-09-03:
Hello,

I just noticed that the combination of both skipping the config mode and having mesh-on-lan enabled in the site.conf, breaks eth0.

The parameter

    network.mesh_lan.auto='1'

is set, but eth0 is not in batctl if and deactivating and reactivating mesh-on-lan multiple times like shown in readthedocs via ssh doesn't fix it, not even with a reboot in between. Even manually adding eth0 with

    batctl if add eth0

doesn't fix it. Tested with two different wr841n v9.

Note: This bug doesn't seem to occur in x86-images. 

Gluon v2016.1.5

Regards,
Matthias

FreifunkUFO, 2016-09-03:
please send output of 
 brctl show

MPW1412, 2016-09-04:
Hello,

I just noticed, that someone unplugged the cable. But I think this should make a difference, right?

```
# swconfig dev switch0 show
Global attributes:
	enable_vlan: 1
Port 0:
	pvid: 1
	link: port:0 link:up speed:1000baseT full-duplex txflow rxflow 
Port 1:
	pvid: 1
	link: port:1 link:down
Port 2:
	pvid: 1
	link: port:2 link:down
Port 3:
	pvid: 1
	link: port:3 link:down
Port 4:
	pvid: 1
	link: port:4 link:down
VLAN 1:
	vid: 1
	ports: 0 1 2 3 4 
# brctl show
bridge name	bridge id		STP enabled	interfaces
br-wan		7fff.16cd20709f02	no		eth1
br-client		7fff.14cc20709f02	no		bat0
# batctl if
mesh-vpn: active
ibss0: active
# uci show|grep mesh|grep lan
network.mesh_lan=interface
network.mesh_lan.ifname='eth0'
network.mesh_lan.auto='1'
network.mesh_lan.macaddr='16:cd:21:70:9f:02'
network.mesh_lan.mesh='bat0'
network.mesh_lan.proto='batadv'
```

NeoRaider, 2016-09-04:
netifd will only setup interfaces that have a link. When the cable is unplugged, the interface won't be added to batman-adv.

NeoRaider, 2016-09-05:
As the cable was unplugged, this sounds like everything is working as indended.

closed 2016-09-05 by NeoRaider

MPW1412, 2016-09-05:
Okay, thanks for the information. I didn't know that, so I guess everything's just fine here.