freifunk-gluon/gluon #880: fastd socket not writeable (and therefore not readable)

viisauksena, 2016-09-10:
a socket have to be writeable to read from it .. 
(took me quiet a while to find this out ...) 

the explanation on https://forum.freifunk.net/t/connect-fastd-socket-on-tp841v11-fastd-statistiken/13499
 to connect will fail if socket is not writeable

therefore user/ownership have to be changed, or it have to be world writeable

NeoRaider, 2016-09-10:
This ticket is missing essential information: What are you trying to do with the status socket?

Changing the permissions of the socket is not recommended. We might want to add better privilege levels in the future, for example services running as "nobody" that should not be able to connect to the socket, but for now, almost everything runs as root, so just use root for your service as well.

viisauksena, 2016-09-10:
your right , i made some other mistake so this shortly didnt work, 
i just wanted to read statistics from socket
(while generaly this is true - it is nothing big since nearby everything run as root on the small dev. like you said - so i close this)

closed 2016-09-10 by viisauksena