freifunk-gluon/gluon #881: cpe210 v1 v1.1 - masterbranch failt

viisauksena, 2016-09-12:
ich habe mit unserem letzten update alle betroffenen cpe210 v1 und v1.1 abgeschossen .. symptom : Geräte tot, leuchtet nur noch power - meist nur config mode
der upgrade passierte von
v2016.1.5.u / gluon-v2016.1-183-gfb8c7a8
auf
v2016.1.6 / gluon-v2016.1-295-gc673356
mit den 841 egal welcher art gibt es keine Probleme mit der FW ... 
sysupgrade oder autoupdater verhalten sich indifferent (logisch)
factory/sysupgrade FW sind ebenfalls indifferent

kb-light, 2016-09-12:
Also fb8c7a8 9855789 scheint es als commit zu geben, c673356 und 655d1ce allerdings nicht. Auch kann ich deinen Versionsangaben nicht ganz folgen, denn v2016.1.6 gäbe es ja nur im Branch v2016.1.x, wobei die Commits scheinbar aus dem master stammen. Nutzt du/ihr direkt das gluon repository oder habt ihr noch ein eigenes mit zusätzlichen Patches drin?

NeoRaider, 2016-09-12:
Joa, die Versionsnummern 2016.1.5 und 2016.1.6 für Mastercommits sind natürlich extrem unglücklich gewählt, da wir dieselben Nummern in Gluon für völlig andere Versionen verwenden... naja, egal.

@viisauksena Hast du verifiziert, dass ein aktueller Build nach `make clean` dasselbe Problem aufweist? Zwischen den von dir angegebenen Commits gab es eine Änderung an den Images für die CPE210, die zu genau diesem Problem führen würde, wenn der Kernel nicht neu gebaut wurde.

viisauksena, 2016-09-13:
@NeoRaider genau das test ich noch ... werd das morgen mal probeweise flashen

viisauksena, 2016-09-14:
getestet an 2 cpe mit sysupgrade scheint das der Fall gewsen zu sein (make clean issue)

@NeoRaider kannst du das genauer sagen welche Änderung das war? 

und vielleicht sollte man neben dem make update (was ich regulär in mein build script aufgenommen habe) auch ein make clean in der Readme.md empfehlen (wenn man vorher schon mal gebaut hat) oder das irgendwie anders promiment platzieren (viell. bei dem entsprechenden commit) - denke ich bin nicht der erste der in diese Falle getappt ist (mir ist, als sei das 2-3 mal Thema im Forum gewesen)

closed 2016-09-14 by viisauksena