freifunk-gluon/gluon #887: dhcp-client on WAN does not send hostname

tobox, 2016-09-22:
With 2016.2, my nodes do not send the desired hostname in the dhcp request on WAN *verified with wireshark, option 12 is empty). Combined with changing MAC addresses on WAN, this makes it difficult to discover the devices, because they get new IP-Adresses without hostnames.

The hostname was plain ascii, no unicode involved.

Happened both after an upgrade from 2016.1.6 and with a clean install.

NeoRaider, 2016-09-23:
Interesting. Is the hostname set correctly in `/etc/config/system` in the `hostname` option of the system section?

tobox, 2016-09-23:
Yes, I would say so:

config system
        option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option hostname 'Bensheim-006'

tobox, 2016-09-23:
If I add `option hostname 'bensheim-002'` to `config interface 'wan'` in `/etc/config/network`, ddns works. But that is just an ugly workaround.

Delta1977, 2016-09-27:
after uci set network.wan.hostname='thehostname'  and "uci commit" the router send his hostname in the request

NeoRaider, 2016-10-01:
I just checked with Gluon 2016.1.6, and it doesn't send a hostname either.

Delta1977, 2016-10-01:
I use Gluon since the first days and the Node never send his hostename to dhcp Server. After the command above he send it. Tested with 2016.1.5. i think this is an openwrt Problem.

NeoRaider, 2016-10-01:
This is not a problem, but a design decision.

Adorfer, 2016-10-02:
Then i'd like to have in configurable in the advanced/network settings. 
(Yes, there are scenarios where you like to have a gluon router go 'undercover'. But that's probably not the default usecase.)

CodeFetch, 2019-02-24:
This was fixed in https://github.com/openwrt/openwrt/commit/abdf0dea3ab6470b596096635a60fbe2e8d0dfff
I would not like to support a feature to hide the hostname, but close this issue. If someone has a creepy setup, he or she should set a random hostname for the WAN-device manually.

rotanid, 2019-02-24:
thanks for pointing this out, so it is fixed in Gluon 2018.2

closed 2019-02-24 by rotanid