freifunk-gluon/gluon #889: ibss0 dies sometimes - ath: phy0: Timeout while waiting for nf to load

viisauksena, 2016-09-25:
we observe a raising fail of ibss0 if - we guess since we have primary0 if 
this means meshing nodes cut themselve off the net and wont come back
since there is no restart logic

```
[32065.740000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
[34704.030000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
[41519.760000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
[46558.520000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
```

viisauksena, 2016-09-26:
ein weiterer dmesg mit mehr Info dazu 
```
[   34.680000] random: nonblocking pool is initialized
[ 6848.890000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
[20915.890000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda
[21608.080000] ------------[ cut here ]------------
[21608.090000] WARNING: CPU: 0 PID: 7386 at /media/freifunk/firmware/gluon/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/linux-ar71xx_generic/compat-wireless-2016-06-20/net/mac80211/sta_info.c:450 sta_info_insert_rcu+0xb0/0x578 [mac80211]()
[21608.110000] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat sch_fq em_cmp sch_teql em_nbyte sch_dsmark sch_pie act_ipt sch_gred em_meta sch_htb em_text sch_codel sch_red sch_sfq sch_prio act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc batman_adv libcrc32c ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables dummy tun act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug crc16 crc32c_generic crypto_hash
[21608.220000] CPU: 0 PID: 7386 Comm: kworker/u2:1 Not tainted 3.18.29 #29
[21608.230000] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[21608.230000] Stack : 00000001 00000000 815d8e80 80359f0c 80e577f4 80e04e48 000001c2 80e128f0
[21608.230000] 	  81804a10 81804a00 00000088 800a3d80 00001cda 80096990 81662908 80e128f0
[21608.230000] 	  8036ab5c 8151fcbc 8151fcbc 800a3d80 00000003 80094380 000001c2 00000000
[21608.230000] 	  00000000 801d99e4 00000000 00000000 00000000 00000000 00000000 00000000
[21608.230000] 	  70687930 00000000 00000000 00000000 00000000 00000000 80ca4800 80ca4900
[21608.230000] 	  ...
[21608.270000] Call Trace:
[21608.270000] [<8007174c>] show_stack+0x50/0x84
[21608.280000] [<80080c60>] warn_slowpath_common+0x84/0xb4
[21608.280000] [<80080d18>] warn_slowpath_null+0x18/0x24
[21608.290000] [<80e04e48>] sta_info_insert_rcu+0xb0/0x578 [mac80211]
[21608.300000] [<80e112a4>] ieee80211_get_vht_mask_from_cap+0xc90/0x1b14 [mac80211]
[21608.300000] 
[21608.310000] ---[ end trace 39a0c4c941ce9b43 ]---
```
ich habe danach noch einiges an 3 Testroutern gesetzt in der Hoffnung noch detailiertere Info zu bekommen.  - dazu **logread -f**
```
# set some sysctl
sysctl -w vm.panic_on_oom=0 # deact reboot on oom
sysctl -w kernel.panic=80 # 80s after panic reboot
sysctl -w vm.overcommit_memory=2 # calc if enough mem is avail mmaloc

# raise prob. of proc to kill
# echo 10 > /proc/$(cat /var/run/fastd.mesh_vpn.pid)/oom_adj # deprecated
echo 700 > /proc/$(pgrep fastd)/oom_score_adj # fastd
echo 900 > /proc/$(pgrep ntp)/oom_score_adj # ntp
echo 950 > /proc/$(pgrep /usr/sbin/batadv-vis)/oom_score_adj # batvis
```

kpanic23, 2016-09-27:
Actually the source of this problem seems to be https://bugs.lede-project.org/index.php?do=details&task_id=176
At least @NeoRaider told me so via IRC on friday. 
A fixed 2016.2.1 is on the way.

viisauksena, 2016-09-27:
@kpanic23 you wont believe how happy this message make me ;)
in the meantime i wrote a emergence script and use this in our network - this also gives plenty more kernel warning (because of sysctl tweaking) ... i add them after the script which somebody find usefull in this or other cases ... 

**emergency script run via micron d every minute**
```
$ cat emergency.sh 
#!/bin/sh

# set some sysctl
sysctl -w vm.panic_on_oom=0 # deact reboot on oom
sysctl -w kernel.panic=80 # 80s after panic reboot
sysctl -w vm.overcommit_memory=2 # calc if enough mem is avail mmaloc

# raise prob. of proc to kill
# echo 10 > /proc/$(cat /var/run/fastd.mesh_vpn.pid)/oom_adj # deprecated
echo 700 > /proc/$(pgrep fastd)/oom_score_adj # fastd
echo 900 > /proc/$(pgrep ntp)/oom_score_adj # ntp
echo 950 > /proc/$(pgrep /usr/sbin/batadv-vis)/oom_score_adj # batvis

# if we see bat GW just exit
netz=$(batctl gwl -H|wc -l)
if [ $netz -ne 0 ] ; then 
        echo "$0 found GW in network, exiting"|logger
        echo 0 > /tmp/emergency
        exit 0
fi

# see ath9k for stopped
# cat /sys/kernel/debug/ieee80211/phy0/ath9k/queues

# simple counter 
touch /tmp/emergency
counter=$(cat /tmp/emergency)
if [ -z $counter ] ; then counter=0 ; fi
if [ $counter -lt 10 ]
        then 
		let counter+=1
		echo $counter > /tmp/emergency
		if [ $counter -eq 5 ]; then echo "$0 - 5 min offline - try wifi"|logger; wifi ; fi
		if [ $counter -eq 7 ]; then echo "$0 - 7 min offline - try restart fastd"|logger; /etc/init.d/fastd restart ; fi
        else reboot
fi 
echo $counter
```

**dmesg after triggered wifi**
```
[    0.000000] Linux version 3.18.29 (fffr@v32412.1blu.de) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r49261) ) #8 Tue Sep 27 05:16:43 CEST 2016
[    0.000000] MyLoader: sysp=37edff37, boardp=8b7fbfcf, parts=2dbf0f19
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9533 ver 1 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x01ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x01ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x01ffffff]
[    0.000000] On node 0 totalpages: 8192
[    0.000000] free_area_init_node: node 0, pgdat 803b6a30, node_mem_map 81000000
[    0.000000]   Normal zone: 64 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[... 480 lines cut ...]
[45027.190000] Mem-Info:
[45027.200000] Normal per-cpu:
[45027.200000] CPU    0: hi:    0, btch:   1 usd:   0
[45027.200000] active_anon:940 inactive_anon:32 isolated_anon:0
[45027.200000]  active_file:1255 inactive_file:564 isolated_file:0
[45027.200000]  unevictable:0 dirty:0 writeback:0 unstable:0
[45027.200000]  free:466 slab_reclaimable:413 slab_unreclaimable:1588
[45027.200000]  mapped:521 shmem:62 pagetables:100 bounce:0
[45027.200000]  free_cma:0
[45027.240000] Normal free:1864kB min:668kB low:832kB high:1000kB active_anon:3760kB inactive_anon:128kB active_file:5020kB inactive_file:2256kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:28308kB mlocked:0kB dirty:0kB writeback:0kB mapped:2084kB shmem:248kB slab_reclaimable:1652kB slab_unreclaimable:6352kB kernel_stack:408kB pagetables:400kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[45027.280000] lowmem_reserve[]: 0 0
[45027.280000] Normal: 220*4kB (EMR) 115*8kB (EMR) 0*16kB 0*32kB 1*64kB (R) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 1864kB
[45027.290000] 1881 total pagecache pages
[45027.300000] 0 pages in swap cache
[45027.300000] Swap cache stats: add 0, delete 0, find 0/0
[45027.310000] Free swap  = 0kB
[45027.310000] Total swap = 0kB
[45027.310000] 8192 pages RAM
[45027.320000] 0 pages HighMem/MovableOnly
[45027.320000] 1115 pages reserved
[45027.340000] br-client: port 3(client0) entered forwarding state
[45027.370000] IPv6: ADDRCONF(NETDEV_CHANGE): ibss0: link becomes ready
[45027.920000] batman_adv: bat0: Adding interface: ibss0
[45027.930000] batman_adv: bat0: Interface activated: ibss0
```

viisauksena, 2016-09-30:
can we expect that this patch is coming into the lede branch - which hopefully will be out this weekend?  or will it be backportet .. 
- beside freiburg (fffr) ffrn seem to have similar problems , actually i dont know any community which [tests 2016.2](https://forum.freifunk.net/t/gluon-v2016-2-released/13575/24?u=fuzzle)
https://bugs.lede-project.org/index.php?do=details&task_id=176

NeoRaider, 2016-09-30:
I have backported the fix and added it to the master and v2016.2.x branches.

viisauksena, 2016-10-01:
i think it is solved, i tried on 3 routers (different - uplink, meshing, heavy load - all 841- with 32mb ram) and they run for 6 hours without any problem 

some things make me curious , 
what this primary0 if is for?
what this multicast hash table is for, what does disabling snooping on bat0 mean in this case

thx a lot for taking care 
```
[   35.990000] batman_adv: bat0: Adding interface: primary0
[   35.990000] batman_adv: bat0: Interface activated: primary0
[   36.000000] device bat0 entered promiscuous mode
[   36.010000] br-client: port 3(bat0) entered forwarding state
[   36.010000] br-client: port 3(bat0) entered forwarding state
[   36.020000] batman_adv: bat0: Adding interface: mesh-vpn
[   36.030000] batman_adv: bat0: The MTU of interface mesh-vpn is too small (1280) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1528 would solve the problem.
[   36.060000] batman_adv: bat0: Interface activated: mesh-vpn
[   36.100000] batman_adv: bat0: no_rebroadcast: Changing from: disabled to: enabled
[   37.050000] br-client: port 2(client0) entered forwarding state
[   37.990000] batman_adv: bat0: Changing gw mode from: off to: client
[   38.010000] br-client: port 3(bat0) entered forwarding state
[   38.050000] batman_adv: bat0: hop_penalty: Changing from: 30 to: 15
[   38.050000] batman_adv: bat0: orig_interval: Changing from: 1000 to: 5000
[   38.200000] batman_adv: bat0: Adding interface: ibss0
[   38.210000] batman_adv: bat0: Interface activated: ibss0
[   40.080000] random: nonblocking pool is initialized
[   48.680000] icmp6_send: no reply to icmp error
[   48.680000] icmp6_send: no reply to icmp error
[   48.680000] icmp6_send: no reply to icmp error
[   48.680000] icmp6_send: no reply to icmp error
[   48.680000] icmp6_send: no reply to icmp error
[   48.680000] icmp6_send: no reply to icmp error
[   70.280000] br-client: Multicast hash table maximum of 512 reached, disabling snooping: bat0
```

viisauksena, 2016-10-01:
and while this may was a serious issue, the bugfix should also affect the tag 2016.2 - since this bug make these builds pretty unusable  (if i am right)

NeoRaider, 2016-10-01:
@viisauksena, it would really help if you'd not ask off-topic questions in the bug tracker, but on the mailing list or in IRC.

`primary0` is a dummy interface that is always added to bat0 first to set the primary batman-adv address as described in http://gluon.readthedocs.io/en/latest/dev/mac_addresses.html (batman-adv doesn't have another way to control the primary MAC address, it will always choose the address of the first added interface).

The multicast hash table is used by the bridge br-client to keep information about multicast groups behind the different bridge ports. It would actually be desirable to disable the snooping on bat0 from the beginning (there are just too many clients "behind" bat0, it doesn't make sense to keep track of all of them), but the kernel doesn't have such a feature at the moment. Thus the kernel will automatically disable the snooping when it detects the high number of multicast groups; this is fine, but will cause the warning message you noticed.

The bugfix has already been backported to the v2016.2.x branch and will be included in v2016.2.1.

viisauksena, 2016-10-01:
many hours later - it seemed that the problem is still there, but is triggered far less often ... so nodes running longer for a while .. but on one i had a runing **logread** and got this 
```
Sat Oct  1 03:31:01 2016 user.notice root: /lib/gluon/ssid-changer/emergency.sh found GW in network, exiting
Sat Oct  1 03:31:01 2016 user.notice root: /lib/gluon/ssid-changer/ssid-changer.sh - still on freiburg.freifunk.net
Sat Oct  1 03:31:25 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.debug kernel: [ 6241.610000] ieee80211 phy0: failed to copy skb for ibss0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] ksoftirqd/0: page allocation failure: order:0, mode:0x20
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] CPU: 0 PID: 3 Comm: ksoftirqd/0 Not tainted 3.18.29 #1
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] Stack : 00000000 00000000 00000000 00000000 80414292 00000036 00000000 00000000
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 	  00000000 00000000 80359f6c 803b7323 00000003 804134d0 81828bf8 00000000
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 	  00000000 00000000 803b6f7c 800a3d80 00000006 80080b58 00000000 00000000
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 	  8035d564 8183bbdc 00000000 00000000 00000000 00000000 00000000 00000000
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 	  ...
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] Call Trace:
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<8007174c>] show_stack+0x50/0x84
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<800cd66c>] warn_alloc_failed+0xe8/0x10c
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<800cf55c>] __alloc_pages_nodemask+0x51c/0x68c
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<8022a908>] __netdev_alloc_frag+0x78/0x1c8
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<8022fe1c>] __netdev_alloc_skb+0x5c/0x118
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80e70090>] ath_rxbuf_alloc+0x30/0x98 [ath]
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80ee70f8>] ath_rx_tasklet+0x588/0xb18 [ath9k]
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80ee4ca4>] ath9k_tasklet+0x23c/0x2c8 [ath9k]
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80083850>] tasklet_action+0x84/0xcc
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80083160>] __do_softirq+0xf8/0x230
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<800832c0>] run_ksoftirqd+0x28/0x58
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80099104>] smpboot_thread_fn+0x134/0x164
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<8009663c>] kthread+0xd8/0xe4
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] [<80060878>] ret_from_kernel_thread+0x14/0x1c
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] 
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] Mem-Info:
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] Normal per-cpu:
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] CPU    0: hi:    0, btch:   1 usd:   0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] active_anon:938 inactive_anon:33 isolated_anon:0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000]  active_file:293 inactive_file:322 isolated_file:0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000]  unevictable:0 dirty:0 writeback:0 unstable:0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000]  free:63 slab_reclaimable:401 slab_unreclaimable:3436
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000]  mapped:387 shmem:63 pagetables:100 bounce:0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000]  free_cma:0
Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] Normal free:252kB min:668kB low:832kB high:1000kB active_anon:3752kB inactive_anon:132kB active_file:1172kB inactive_file:1288kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:28308kB mlocked:0kB dirty:0kB writeback:0kB mapped:1548kB shmem:252kB slab_reclaimable:1604kB slab_unreclaimable:13744kB kernel_stack:368kB pagetables:400kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable?Sat Oct  1 03:31:30 2016 kern.warn kernel: [ 6241.620000] lowmem_reserve[]: 0 0
[... cut ...]
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.740000] ath: skbuff alloc of size 1974 failed
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.750000] ath: skbuff alloc of size 1974 failed
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.750000] ath: skbuff alloc of size 1974 failed
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.760000] ath: skbuff alloc of size 1974 failed
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.760000] ath: skbuff alloc of size 1974 failed
Sat Oct  1 03:31:30 2016 kern.err kernel: [ 6249.770000] ath: skbuff alloc of size 1974 failed

```

NeoRaider, 2016-10-01:
Your log shows a simple out-of-memory situation, it has nothing to do with this bug.

NeoRaider, 2016-10-01:
Nevertheless, it seems there are still ath9 stability issues.

Casey1979, 2016-10-01:
viisauksena is suffering from the same bug as me. https://github.com/freifunk-gluon/gluon/issues/813
And the issue there is that the kernel itself allocates skbs until it is oom, limited only by the size of its adress space. At least that is how far i came with my investigations. Neither sysctls nor a switch from ibss to mesh or a reduction of ath9k receive buffers does help.
The less cpupower and ram the bigger the issue when the wifi interface gets flooded.
Something like a ringbuffer that deallocates unreceived skbs in overrun conditions might help but i am not capable of doing that.

viisauksena, 2016-10-02:
on some node (seems with more ibss0 links they crash more often - and from those more often on daytime (when used))
(this is a list with latest master FW, and the newest flashed ones are 18h ago)
[image]

viisauksena, 2016-10-12:
i build with latest code, the reboots still are there .. dont know if the reason is the same
https://github.com/freifunk-gluon/gluon/commit/487922a8d9b7ac564effe25632fd90e2cb6979b9

viisauksena, 2016-11-05:
so, with latest source there is no longer this oom killed stuff, so not directly related to ibss0 dies sometimes, on very rare occurencys we observe that network ggoes crazy, all wifi seem down, interfaces no longer exist - something like @Adorfer mentioned sometimes..
but this all is very confusing. 
fact is, that we have heavy reboot on crowded mesh island (still bat-v14) - so i am unsure to close this here - and open a new topic ... but calling this what?
to get an idea of the problem - open meshviewer and sort node list by links and see uptime.
there are no entries in dmesg or syslog , even if watched remotely they simply reboot at a certain point ... 
[image]

rotanid, 2016-11-13:
let's continue with this in the main issue regarding ath9k problems: #605

closed 2016-11-13 by rotanid