freifunk-gluon/gluon #897: cpe210v1 rssi leds using 12%cpu

viisauksena, 2016-10-03:
(rssileds is the feature which show rx level of ibss0 meshing network on the small leds on the side of cpe210  -  as far as i know)  

i suggest to let rssi leds only run for the first 15 minutes, or same other reasonable time ... so the 12% cpu is free and available for fastd or other things 

```
Mem: 28276K used, 32520K free, 252K shrd, 2216K buff, 7316K cached
CPU:  20% usr  26% sys   0% nic  16% idle   0% io   0% irq  36% sirq
Load average: 1.86 1.97 1.90 1/51 9861
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1644     1 root     R     3408   6%  49% /usr/bin/fastd --config - --daemon --
 1751     1 root     R     1216   2%  12% /usr/sbin/rssileds ibss0 40000 1 tp-l
 1606     1 root     R     1532   3%   5% /usr/sbin/hostapd -P /var/run/wifi-ph
```

mweinelt, 2016-10-03:
Or maybe just let the program update only in intervals, instead of hammering them so much that they flicker.

TobleMiner, 2016-11-02:
Agreed. But this should be fixed in [openwrt itself](https://github.com/openwrt/openwrt/blob/cabb0c506526fe468e4aa985086e50c440192122/target/linux/ar71xx/base-files/etc/board.d/01_leds#L134) The default value is 40ms (25 refreshes per second) which is really fast. I think that 250 ms (4 refreshes per second) would be much more reasonable. I would suggest reopening this issue on the openwrt repo.

viisauksena, 2016-11-02:
but this is also installed by default in gluon for cpe210 at least - so limiting rssileds in gluon in whatever-way  also sounds reasonable

TobleMiner, 2016-11-02:
Sure, gluon should fix this. But it is also "broken" in openwrt. Thus it should be fixed there, too.

mweinelt, 2016-11-02:
So can someone confirm 250ms to be a good value?

viisauksena, 2016-11-05:
so we wait until PR in openwrt is acknowledged and than build it in gluon (and whats than with upcoming LEDE) or we need own patch?

NeoRaider, 2016-11-05:
We'll need a patch in Gluon as well.

@mweinelt, I think it would make sense to change the interval for all devices at once, not just for the CPE210/510.

mweinelt, 2016-11-05:
So set 200ms for every device that has less than 200ms?

NeoRaider, 2016-11-05:
Yes. A quick grep shows that all devices use the same interval of 40ms at the moment.

viisauksena, 2016-11-12:
is there a  place where i can change it once , or do i have to change all the profiles with a patch?

commit 9279fe4b (2016-11-16): ar71xx, ramips: reduce CPU load and flickering on devices using rsslieds

Fixes #897
commit 9c0a0898 (2016-11-16): ar71xx, ramips: reduce CPU load and flickering on devices using rsslieds

Fixes #897
closed 2016-11-16 by NeoRaider