freifunk-gluon/gluon #898: wrong version on status page for TL-WR941ND Version 3.6

steinjer, 2016-10-03:
on the status page of the node is v2 shown, but the hardware is version 3.6

> cat /tmp/sysinfo/model
> TP-LINK TL-WR941N/ND v2
>
> cat /proc/cpuinfo
>Atheros AR9132 rev 2
>TP-LINK TL-WR941ND 

could it be a problem of openwrt? In the makefile (trunk/target/linux/ar71xx/image/Makefile) have both the HWREV = 2 and the same HWID:

    913 define Device/tl-wr941nd-v2
    914 $(Device/tplink-4m)
    915 BOARDNAME := TL-WR941ND
    916 DEVICE_PROFILE := TLWR941
    917 TPLINK_HWID := 0x09410002
    918 TPLINK_HWREV := 2
    919 endef
    920
    921 define Device/tl-wr941nd-v3
    922 $(Device/tplink-4m)
    923 BOARDNAME := TL-WR941ND
    924 DEVICE_PROFILE := TLWR941
    925 TPLINK_HWID := 0x09410002
    926 TPLINK_HWREV := 2

> cat /proc/mtd
>dev: size erasesize name
>mtd0: 00020000 00010000 "u-boot"
>mtd1: 00120e00 00010000 "kernel"
>mtd2: 002af200 00010000 "rootfs"
>mtd3: 00060000 00010000 "rootfs_data"
>mtd4: 00010000 00010000 "art"
>mtd5: 003d0000 00010000 "firmware"

I've attached the art.dump:

[art_941.zip](https://github.com/freifunk-gluon/gluon/files/506520/art_941.zip)

NeoRaider, 2016-10-03:
There is no way to differentiate these devices from the firmware - the hardware is identical.

For this reason, the device will always be shown as a v2 (and either the v2 or the v3 firmware may be flashed, the images are the same).

closed 2016-10-03 by NeoRaider

steinjer, 2016-10-04:
Wouldn't it be better to figure this more transparent like this?

/tmp/sysinfo/model
TP-LINK TL-WR941N/ND v2/v3